OctoPrint/OctoPrint #2203: Temperature Offset has to reentered after disconnection of printer

baxerus, 2017-11-03:
#### What were you doing?

1. Printer is connected to OctoPrint (everything is working and printing)
2. Temperature offset (of -30 deg) is set (and working)
3. After prints are ready, printer is switched off (to save power), but OctoPrint is still running
4. After some time (actually the amount of time is irrelevant) the printer is powered on again
5. The connection is automatically reestablished
6. Everything is working (printer is printing...)
7. The temperature offset is still shown as -30 deg (even after a browser reload of the page), BUT is isn't respected when a new print is started
8. To get the temperature offset working again the setup dialog has to be opened and saved again

#### What did you expect to happen?
* Ideally: The still shown temperature offset should still be used, when a printer reconnects to OctoPrint
* Alternatively: The display of the offset temperature is also reset back to zero

#### What happened instead?
* Temperature offset is still shown after reconnecting the printer (even if the browser is reloaded), but is not respected in prints, till it is resaved (after the reconnection).
* It seems like the offset is reset internally, but not in the frontend model.

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint

* OctoPrint 1.3.5 (master branch)
* But the same problem is there at least since the 2 years I'm using it (the rest of OctoPrint is superb by the way 😉 )

#### Operating System running OctoPrint

* OctoPi 0.12.0
* But problem is at least 2 years old (also OctoPi is suberb 😉 )

#### Printer model & used firmware incl. version

* CTC Dual/Bizer (China brand), some sort of Makerbot Replicator clone
* Display says: "Internal Release 003" and on a second line: "Firmware release 7.4". I do not know, what the firmware is based on, but most likely Marlin.

#### Browser and Version of Browser, Operating System running Browser

* Every browser and OS (including mobile) I have been using/tried the last 2 years
* Chrome, Safari
* Mac OS X, Windows, Linux, iOS, Android
* It isn't a client problem

#### Link to octoprint.log

* 3MB Debug logfile: https://gist.github.com/baxerus/00e88658d0726825abcaf926ebf56329
* There are multiple "Connecting" events inside the log, but start looking at the end 😉 

#### Link to contents of terminal tab or serial.log

Terminal tab (from turning off printer and turning it back on again:
'''Unexpected error while writing to serial port: IOError: '[Errno 5] Input/output error' @ comm.py:_do_send_without_checksum:2414
Changing monitoring state from 'Operational' to 'Offline: IOError: '[Errno 5] Input/output error' @ comm.py:_do_send_without_checksum:2414'
Connection closed, closing down monitor
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: <GPX.gpxprinter.GpxPrinter instance at 0x74604cd8>, starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Recv:  Makerbot v7.4
Recv: echo: gcode to x3g translation by GPX
Send: N0 M110 N0*125
Recv: SD init fail
Recv:  T:38 /0 B:72 /0 T0:38 /0 T1:0 /0 @:0 B@:0
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok PROTOCOL_VERSION:0.1 FIRMWARE_NAME:Makerbot FIRMWARE_VERSION:7.4 FIRMWARE_URL:https://support.makerbot.com/learn/earlier-products/replicator-original/updating-firmware-for-the-makerbot-replicator-via-replicatorg_13302 MACHINE_TYPE:r1d EXTRUDER_COUNT:2
Send: M105
Recv: ok T:37 /0 B:71 /0 T0:37 /0 T1:0 /0 @:0 B@:0
Send: M105
Recv: ok T:38 /0 B:71 /0 T0:38 /0 T1:0 /0 @:0 B@:0
Send: M105
'''

#### Link to contents of Javascript console in the browser

* No client problem

#### Screenshot(s)/video(s) showing the problem:

* nope

I have read the FAQ.

foosel, 2017-11-10:
I was a bit unsure which of the two options to solve this to choose. In the end I decided on resetting the offset to 0 on disconnect. The reason for this is that I think it's worse expecting the offset to reset on disconnect and then not having that be the case like it was so far (although the displayed value lied about that) than it would be having to re-enter the offset after a reconnect, considering that offsets in general are more for cases of "oops, I sliced that file with the wrong temperature" and not "let's put a general offset here".

While at it I also made the actual and target temperature reset to 0 on disconnect.

Patch is pushed to `maintenance` and I'll merge it to `devel` on the next opportunity. It will be released as part of 1.3.6.

> the same problem is there at least since the 2 years I'm using it

Why didn't you say something sooner? 😄

baxerus, 2017-11-11:
I would have liked the other option better (because it fit's more for my use case. Totally egoistic 😉 ), but resetting on disconnect is also okay. The most important thing is, the shown value in the frontend is corresponding with what OctoPrint is actually doing 👍 .
I use the feature very often for changing the temperature between PLA, ABS, HIPS whatever or even more often for different brands of ABS, that like other tool temps (and of course for experimenting) without always reslicing.

*Why I didn't say something sooner?* Because it never was such a big deal for me and I hoped, that it would be corrected someday by you anyway.
In the last release then was the new temperature frontend and I was full of hope, but again the problem wasn't solved (by itself 😉 ), so I decided to write the issue. BUT THAN I have seen the issue tracker guidelines... I'm working in the software development business myself and I know why you have/need this long list of thing a user should add to a new issue (it makes the live easier for you), but I really looked into the code first, because finding and fixing the issue myself and adding a MR instead of an issue seemed easier at first 😄 . After looking into the code, I realized, that this wouldn't be that easy also. So I waited some more days and than created the issue...

By the way, I guess you know directly where to search for the initialization after a disconnect and things like that. Where would I start? I already looked (a little bit) into the dev docu, but couldn't find some kind of overview document (best a diagram), describing how the components are put together and interacting (in different events). Was it my fault? Didn't I searched long enough?

baxerus, 2017-11-15:
> To be honest, I never even thought about this particular use case for the offset feature that you are describing there.

I was thinking changing the temperature for different materials (and experimentation) was exactly the idea of that feature 😉 

But yes, perhaps a plugin with the possibility to map temp A to temp B if dropdown C is selected (e.g. dropdown option PLA or ABS) would be the real deal. But much more complicated.

Perhaps I will simply start slicing my files right (but have to do it for different materials again) or use the new GCodeEditor (but the GCodeEditor could benefit with a search and replace feature for this 😉 ).

> Nah, that's more my fault. Some kind of an architectural overview is something that's sitting on my TODO list, staring at me grumpily while it gets overtaken by more pressing issues all the time. I think it has grown somewhat bitter by now and is starting to plan its revenge 😕

As stimulus for a reprioritization:

Perhaps new developers (like me) would more likely start digging into the project and (after some time) start working on shortening your TODO list, if there is a architectural overview 😄 .

And it is also always a good start for revisiting everything, requisitioning it and getting new insights into your own old project 😉  (I'm the motivation master!).

closed 2017-12-12 by foosel

wescrockett, 2018-04-12:
Could this be a configurable preference?

foosel, 2018-04-12:
Sure, send a PR.

commit a6eac391 (2017-11-10): Reset temperature offsets on disconnect

Fixes #2203