OctoPrint/OctoPrint #2206: Disabling Webcam not possible

mgrl, 2017-11-04:
#### What were you doing?

Earlier I had a webcam installed, which worked. Now I would like to disable related settings. Have done this by removing all URLs in the webcam-settings tab.

#### What did you expect to happen?
No capturing from webcam any more.

#### What happened instead?
Octoprint tries further to get a stream at every print started, which is not possible, so I get error messages at every print. It seems, the settings are not translatet correctly to the config.yaml? There I have:

webcam:
  ffmpeg: ''
  snapshot: ''
  stream: ''
  timelapse:
    fps: 25
    options:
      capturePostRoll: true
      interval: 2
    postRoll: 0
    type: timed
  watermark: false

Guess it should have something like {} in there?
Here is a log-entry related to that:
2017-11-03 20:52:30,230 - octoprint.timelapse - ERROR - Could not capture image /home/pi/.octoprint/timelapse/tmp/xxx-0.jpg from 
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/timelapse.py", line 559, in _perform_capture
    r = requests.get(self._snapshot_url, stream=True, timeout=5)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/api.py", line 69, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/api.py", line 50, in request
    response = session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/sessions.py", line 451, in request
    prep = self.prepare_request(req)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/sessions.py", line 382, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/models.py", line 304, in prepare
    self.prepare_url(url, params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/models.py", line 362, in prepare_url
    to_native_string(url, 'utf8')))
MissingSchema: Invalid URL '': No schema supplied. Perhaps you meant http://?

#### Branch & Commit or Version of OctoPrint

1.3.5

#### Operating System running OctoPrint

OctoPi 0.13

#### Printer model & used firmware incl. version

Marlin 1.1.x

Can anybody reproduce that?

foosel, 2017-11-06:
To be honest, I never anticipated someone disabling timelapse in general by unconfiguring the urls and ffmpeg but leaving the timelapse settings intact (in your case timed timelapse at 2s interval) and the code simply didn't handle that correctly.

I just pushed the above commit that will rectify such situations and set timelapse to "off" if snapshot url or ffmpeg path are unset. That should fix situations like this in the future. Will be part of 1.3.6 (already available on `maintenance` and soon `devel`).

In the meantime, you have two options to fix this on 1.3.5 and earlier: 

  * Either set the snapshot url and ffmpeg path back to something, reload, open the now again available timelapse tab and set Timelapse mode to "off" and save that as default, then unset your snapshot url and ffmpeg path again. 
  * Or edit your `config.yaml` and remove the line that says `timelapse: timed` - that will reset it to "off" and do basically the same as the option above.

---

In the future, **please fill out the full ticket template**. Thankfully this issue was solvable with the limited amount of information you provided, but that might have not been the case if the error you pasted wasn't immediately recognizable as a symptom of the actual underlying problem, and in that case it would not have been possible to look into this in a timely manner. Running after information needed to solve an issue is sometimes more work than solving the issue once it has been understood and takes away valuable time to could better be used to actually add value to the software, and the template asks for logs and such in order to avoid this as far as possible.

mgrl, 2017-11-06:
@foosel thank you very much for fixing that.

> Either set the snapshot url and ffmpeg path back to something, reload, open the now again available timelapse tab and set Timelapse mode to "off" and save that as default, then unset your snapshot url and ffmpeg path again.

Indeed I was looking for some month now (ok, always when the popups began to nag again) for that tab to configure these settings and had no idea where to find that... Everything clear now...

commit 5aad696c (2017-11-06): Do not enable timelapse if settings are missing

If snapshot url and/or ffmpeg path are not set, the timelapse cannot
work regardless of what type has been configured.

So if the current configuration still contains a timelapse type other
than "off" but snapshot url or ffmpeg path are unset, force a disabled
timelapse.

Fixes #2206
closed 2017-11-06 by mgrl