OctoPrint/OctoPrint #2209: [Request]: Warn user if gcodes file seems not to be complete

Boby71, 2017-11-05:
Currently, I was hunting down problems assuming the serial connection between my RPi3 and my Prusa I3 MK2s were having problems: [https://github.com/prusa3d/Prusa-Firmware/issues/234](url)

But I learned, that my gcode file got cut off in the middle while writing to the uploads directory of Octoprint. This error is a kind of dangerous: The movement of the printer stops and as there are no commands to lower the temperatures, the printer kept the bed and the nozzle heated up until I manually lowered the temperatures.

In the latest firmware Prusa introduced a check (when printing from SD card), if the file is complete. It's a simple check:
```
**Check if file is complete**

If file is not complete, user is informed by warning message about it and he can decide if he will continue printing anyway. If user choosed not to print incomplete file, there was bug which caused, that printer executed some movements. This has been fixed.

For this check we search for M84 (disable steppers) in last 10 000 bytes of gcode file. This works well with our print settings. If you don't use our print settings, please add M84 to end section of your gcode.
```
I'd like Octoprint to check the same - if parts of the typical ending scripts are missing (= one or all of these lines: 
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M84 ; disable motors
), then Octoprint should warn the user to check if the file is complete.

ntoff, 2017-11-05:
Why not just use M02? That's literally what it's for.

http://reprap.org/wiki/G-code#M2:_Program_End

From the engineering book in front of me (not the reprap wiki)

>M02 
>At the completion of the machining operation, this end-of-program code stops the machine when all commands in the block are executed. May include rewinding the tape.

So while it's not specifically supported by most firmwares, if OctoPrint were to search for that, it'd be more appropriate than "M84" since not everyone may want their motors disabled at the end, and I think maybe was it @nophead? even leaves his heaters on if printing multiple items so you couldn't *just* rely on searching for any "heaters off" commands either.

You may want to make this a user configurable search term.

Boby71, 2017-11-05:
I just gave some possible examples; maybe they are not valid for everyone. A configurable check item would be geat (could also be a comment being added by the slicer at the end of the generated gcode).

Someone else gave me an idea for a nice workaround: Put the commands for lowering temps and disable steppers in the „after print script“ of Octoprint. Still my print is failed at that point, but it’s no safety issue anymore.

nophead, 2017-11-06:
Not me, I always turn my heaters off and cool the bed with a fan and wait for it to cool before turning the fan off.

ntoff, 2017-11-06:
oh ok, I could have sworn I saw someone (thought it was you) say they kept the heaters on between prints and just swap out the bed or something.

Still, my point about people doing that stands, sorry to drag you into this. :(

Boby71, 2018-11-19:
Had a corrupted GCODE again; I still think there should be an option in Octoprint to screen the GCODE for a certain text, e.g. in a comment. I can add this comment in my slicer in the "end-of-gcode" section. By default this option could be deactivated, but as soon as it's manually activated, Octorpint starts only to print if the text has been found.