OctoPrint/OctoPrint #2211: python script as custom control

hexxter, 2017-11-05:
1. Version 1.3.5 https://github.com/foosel/OctoPrint/commit/77753ca02602d3a798d6b0a22535e6fd69ff448a
2. feature request
3. reason for this request: its a good thing to control the printer with the power of a python script. In my case is it to drive my toolchanger (pick z probe or nozzle) and measure the distance between z probe and my nozzle on a touch plate. After the measurments i have to Z values from the printer and will set the differenz as nozzle offset the the printer via GCODE.

What is my issue:
on a custom control i will run a python script what can fire gcode to the printer and get results (answers) from the printer.

I think this is a good way to build cool and easy powerfull scripts for OP.

Thanks for your great work!

BR Hexxter

kantlivelong, 2017-11-05:
This sounds like something that should be done as a plugin.

hexxter, 2017-11-05:
i can not imagine to do it with a plugin? If i understand runs the  plugin async and i can only get the result with octoprint.comm.protocol.gcode.sent hook. but i do the same thing with different nozzles or the z probe and in every case i get the same responce "Bed X: 123.13 Y: 123.12 Z: 123.34". In my case the first response is the z probe the next is T0 then T1. 3 Time is the hook triggered but if a error on the printer i dont know and if i make a difference from the wrong Z values and set this as my nozzel offset then crash the printer (nozzle on bed).
I am wrong?

BR Hexxter

foosel, 2017-11-06:
A custom control backed by a python script would give you not more power as having said python script running as part of a plugin 🤔 In fact, via a plugin you'd actually have MORE possibilities. 

What you are looking for would be possible to do with a plugin (with a bit fiddling and working around firmware limitations, which are the actual problem here, not the question of plugin or custom control). Yes, you are limited to sending and receiving lines one by one and having to sort out what means what and what belongs where. But that is ALWAYS the case due to issues with the underlying protocol (e.g. no proper way to reliably associated sent commands with received responses) and nothing that could be solved by what you are asking for here.

Take a look at the octoprint.comm.protocol.gcode.sent and octoprint.comm.protocol.gcode.received hooks and combine that with some kind of internal state machine.

closed 2017-11-06 by foosel