OctoPrint/OctoPrint #2212: Feature request: Long pause.

rewolff, 2017-11-06:
Situation: I have started a long print and now it's over 40% done, I realize that it is going to run out of filament. I did not expect to run out of filament so soon so I don't have a spare. I want to pause the job while waiting for the new filament to arrive. 

For me the "long pause" is: 
* Pause the print
* Move the head up (in my case 100mm is not going to hit the top of my printer, but for other people 20 mm might be more appropriate). 
* set temperatures: OFF
* turn off the motors. 

Of course if this is implemented in octoprint, there needs to be a "resume print". When implemented programmatically you can improve on my procedure by asking the printer where it stopped and starting with a home operation. (oops! Not always possible on [0,0,0] homing printers. It is on my delta!) That would tolerate bumping the printer while paused with the motors off. 

An additional feature might be to save such things like current position in gcode file and current head position so that a restart after a reboot of octoprint or the whole computer might become possible.

foosel, 2017-11-07:
You should be able to already do that with the position information that OctoPrint saves on print pausing and cancelling (and which is also part of the corresponding events and the resume point, see [here](http://docs.octoprint.org/en/master/events/index.html#printing)). See [here](http://docs.octoprint.org/en/master/features/gcode_scripts.html#context) for how that info is available already in pause/resume scripts (heed the warning!) and [here](https://gist.github.com/foosel/1c09e269b1c0bb7a471c20eef50c8d3e) for an example on how to use that for a more sophisticated pause/resume script.

If you don't want to use it as your go-to pause/resume script, using event processing and a custom plugin you could also instead create your own "long pause" button or even generate a resume-gcode-script on pause/cancel.

I think that should already cover what you are looking for here without the need to add any additional things, or am I missing something here?

foosel, 2017-11-20:
Hm... I figure I'm not missing something here, or someone would have said something ;) Closing since already possible with what's there.

closed 2017-11-20 by foosel

rewolff, 2017-11-20:
If it's already possible, that's great for octoprint. But to me it remains a "feature request". 

If you say it can simply be done with a plugin or something like that: I program for a living. I could dive in and make it. But the 3D-printing is a hobby and at the moment I don't have the time to spare to go in and  figure out how to make this. 

If you say it's a simple plugin, how about setting up a repository where people can contribute such plugins and then simply accepting all pull requests? And maybe a mechanism where I can search for such plugins. 

Then your answer to my request would in the future, be as simple as: that's a plugin, search the plugins. Or... "it can be done in a plugin, post a feature/plugin request on the plugins github project".

foosel, 2017-11-20:
Uhm... Might I direct your attention to the [official plugin repository](http://plugins.octoprint.org/) that is also easily accessible (and searchable) right from within the integrated plugin manager? It has been online since around 2014 and people write plugins and send a PRs to get them registered there.

But again, the basic gist of what you asked for doesn't even require a plugin but only adjusted pause and resume scripts (see the links I posted in my first comment).

rewolff, 2017-11-20:
You may. Thank you. :-) :+1: 
F#*$()*@$ stupid users.