OctoPrint/OctoPrint #2213: Temperature graph no longer displays any information

samling, 2017-11-07:
#### What were you doing?

Running a print job via the OctoPrint interface.

#### What did you expect to happen?

I expected the temperature graph to show lines representing the temperature values over time.

#### What happened instead?

The graph momentarily displays the values for actual T and Bed temperatures in the legend, but revert to "-" with no data on the graph, but with values present in the table underneath (where the controls are for setting the temperature).

#### Did the same happen when running OctoPrint in safe mode?

Yes, rebooting in safe mode resulted in no change.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5 (master branch)

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Original Prusa i3 MK2S 3.0.12

#### Browser and Version of Browser, Operating System running Browser

Google Chrome 62.0.3202.75 running on Windows 10

#### Link to octoprint.log

https://gist.github.com/samling/dd51b1f65165d91cc6c078eba4f6ce62

#### Link to contents of terminal tab or serial.log

Link to the terminal tab output after restarting OctoPrint:
https://gist.github.com/samling/8c59e1c471eea3b5fa97b3306de10596

#### Link to contents of Javascript console in the browser

https://gist.github.com/samling/4b30ac699940c818514b66e16cc08a77

#### Screenshot(s)/video(s) showing the problem:

https://i.imgur.com/fazp1KY.gif

kantlivelong, 2017-11-07:
Hello,

What happens if you [run chrome without extensions/plugins](https://superuser.com/questions/822429/how-to-start-chrome-without-plugins) or try another browser?

samling, 2017-11-07:
Hi @kantlivelong, running in Chrome without extensions as well as in Firefox yields the same issue. No change.

foosel, 2017-11-07:
Just an idea (I cannot reproduce this over here and so far also have nothing in the logs spring out at me) but: what happens when you switch to another tab (e.g. Control) and then switch back to the temperature tab? Does it change anything?

Also: What was the last version you saw this working on?

---

And an additional question most likely not related to this particular issue: could you provide the file you were printing when you ran into that one exception in your log (`RuntimeError: dictionary changed size during iteration`)? That looks like it might be the same as #1951, an issue I've been trying to hunt down since a couple months now, so far without any luck or reproduction. So any help there would be greatly appreciated.

samling, 2017-11-07:
@foosel,

Changing the tabs has no effect. I went back and checked the iso I flashed to the raspberry pi SD card and it's 0.14.0, so the same version as I'm on now. For what it's worth, it did work initially, but stopped soon thereafter for no particular reason (that I could think of).

As for the file I was printing, looking in my octoprint history around the time I posted this it looks like I had just finished printing this [calibration cube](https://www.thingiverse.com/thing:1278865) with [this profile](https://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/simplify3d-print-settings-august-30-2016--t938-s260.html#p37358) in Simplify3D. I hope that helps.

foosel, 2017-11-07:
Sorry, I meant the last Octo*Print* version, not the last Octo*Pi* version (the former is the server software running on the latter image). Since Octo*Pi* 0.14 ships with Octo*Print* 1.3.4, I guess that's also the only version you ran before on this instance? Does the "it stopped working" correlate with the update you did to 1.3.5 (which was released on October 16th) or was it before that? 

And thanks for the file and profile. Might not be able to reproduce #1951 with that either, but it's worth a shot :)

samling, 2017-11-07:
Sorry about that, yes, 1.3.4 would be the version I started with. To the best of my recollection it started before that; I seem to remember hoping the update would fix the issue, but it didn't. Apologies for the unsure answers, this has been a problem for at least a month now.

foosel, 2017-11-07:
If you didn't run the release candidates (which you'd actively have had to switch to, which I'm sure you'd remember), this really sounds like an issue that was already present in 1.3.4 for you. What's weird is that it initially worked and then didn't. It could be that something in your config changed maybe but that's grasping after straws 🤔

I have to admit I'm a bit at a loss about this right now, especially since there's no error at all present in the logs you provided that would help narrow this down.

If you have a spare SD card, it might be worth a try to flash a fresh image on that, booting your Pi from it and seeing if you can reproduce.

foosel, 2017-11-10:
Is there a way I could take a look at this instance myself? (port forwarding or alternatively some kind of screen sharing)

samling, 2017-11-10:
@foosel I'm away from home this weekend, but next week I can send you an openvpn profile to connect to my VPN that my Pi running Octoprint is usually connected to. We'll just need to arrange a block of time so I can make sure I'm not printing anything. :)

foosel, 2017-11-13:
Sounds good! You can reach me in private at gina -at- octoprint.org

foosel, 2017-11-20:
@samling Haven't heard from you yet so I guess some stuff got in the way. No problem but: Any estimates? :) Kinda stuck on this until I can take a look I fear, I still haven't seen this myself.

foosel, 2017-12-14:
After a quick skype call it turns out this was caused by the temperature cutoff being set to 0. Going forward OctoPrint will enforce a minimal value of 0 here. Already available on `maintenance`, soon `devel`, will be released with 1.3.7.

foosel, 2018-04-09:
Closing as 1.3.7 was just released.

closed 2018-04-09 by foosel