OctoPrint/OctoPrint #2218: Recv: Error:checksum mismatch during printing on Prusa i3 MMU (3.1.0 RC2)

henrikkoch, 2017-11-12:
the printer stops with heat ON in the print (im using Octoprint 1.3.5. According to the command line in Octoprint it have an error in line 7509 in the included gcode where it says: Recv: Error:checksum mismatch, Last Line: 7509

I here include the STL file, the log from Octoprint and the GCODE file that I did v
[files.to.report.zip](https://github.com/foosel/OctoPrint/files/1465090/files.to.report.zip)
ia SlicerPE and my slicer config bundle

files to report.zip

Hope this helps to find the reason for the error
files.to.report.zip

I used Prusa MMU firmware 3.1.0 RC2  (a newer version has been release today - I don't know if this solves this issue)

henrikkoch, 2017-11-13:
I updated my printer with the new released firmware 3.1.0 and the problem seems to have disappeared. The GCODE file can now be printed without halting. :-)

closed 2017-11-13 by henrikkoch

foosel, 2017-11-13:
Just for the record: The firmware was missing the `ok` there after the `Resend` request, which actually is mandatory. This particular problem however sadly exists in a bunch of firmware variants out there and so OctoPrint even offers a workaround for that. Enabling "Simulate an additional `ok` for resend requests" under Settings > Serial Connection > Advanced options should have done the trick.

henrikkoch, 2017-11-15:
Thanks for that feedback :-)

hoegge, 2018-01-28:
This does not seem to fix the case even with this switched on - it tries a bit more until it stalls completely.