OctoPrint/OctoPrint #2219: Download GCode Link Error with forwarded port

BinoX, 2017-11-13:
#### What were you doing?

When you use a reverse proxy for Octoprint (in my example I am using OctoPi) the links that let you download the gcode don't take the port into account...

e.g. if octoprint is proxied from: http://external.address:8000 the download links for the gcode try and download from http://external.address/downloads/files/local/gcodename.gcode rather than http://external.address:8000/downloads/files/local/gcodename.gcode

I am accessing my octoprint from work and wanted to download a gcode file... My external port for octoprint is 8000

Steps to follow in order to reproduce problem:

1. Set up proxy to forward any port other than 80 to the plain http interface (no tried with SSL).. i.e. port 8000 to octoprint
2. go to octoprint at port 8080
3. try and download a gcode file

[image]

kantlivelong, 2017-11-13:
Hmm... I'm not able to replicate this behavior. How is the proxy configured?

BinoX, 2017-11-15:
I'll post up my config in about a hour. It's an NGINX reverse proxy set up, which I use quite a lot for this kind of task...

I've also noticed that it happens for downloading log files as well, but downloading timelapses is fine

foosel, 2017-11-24:
That sure is one long hour ;)

Any updates on the config?

BinoX, 2017-11-29:
Whoops! Terribly sorry, I completely forgot I posted this... Remembered when I tried to use it again earlier.
I use a similar config for other sites and services and they all work fine... Maybe Octoprint requires some extra settings...

Anyway here is my nginx config:

```
server {
    listen 8000;
    server_name _;

    location / {
        proxy_pass http://192.168.0.34:80;
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
    }
}
```

kantlivelong, 2017-11-30:
Change `proxy_set_header Host $host;` to `proxy_set_header Host $http_host`

foosel, 2017-11-30:
OctoPrint needs to know what port a request is coming in on or it will assume the default port for the URL scheme. For that it will evaluate - in that order - the `X-Forwarded-Host`header, the `Host` header and the `X-Forwarded-Server` and `X-Forwarded-Port` headers.

Nginx' `$host` variable *does never contain the port number* (see also [here](https://stackoverflow.com/a/15414811)). So if you do something like `proxy_set_header Host $host;`, if you connect then via `http://example.com:8080` your nginx will only forward `Host: example.com` and quietly throw away the port number. So for lack of alternative information OctoPrint will default to the scheme default port 80 in such a case. 

`$http_host` on the other hand is the actual `Host` header sent with the request, including the port number (so `example.com:8080`) - so use that and all will work just fine.

OctoPrint does generate some absolute URLs server side based on the incoming request on API responses that are used for downloads and such in the frontend - that's why it also needs the full port information. Other services might only generate relative URLs which then magically work even though the server only gets half of the facts.

BinoX, 2017-11-30:
Whoops, feeling a little embarrassed now. I've been administering Nginx (and various other systems) servers for years and I totally didn't see that I had only put $host instead of $http_host

I should know better than that! Fallen into the "blame everyone except myself" trap. I think I must have used one of my "trick" configs as a template (I have a few services that do not like anything other than the standard port being used... I did not write them!!!!)

Anyway,
Many thanks for showing me my mistake. Donation en route

closed 2017-11-30 by BinoX