OctoPrint/OctoPrint #2221: Could not render movie; OSError: [Errno 20] Not a directory

boschie, 2017-11-14:
#### What were you doing?

Printing any kind of object.

#### What did you expect to happen?
After the print is over it should make a timelapse of the print.

#### What happened instead?
There is an error which is unknown to the system: return code 255.

#### Did the same happen when running OctoPrint in safe mode?
Don't know. I didn't try that.

#### Branch & Commit or Version of OctoPrint
1.3.5 and also 1.3.4.

#### Operating System running OctoPrint
Raspbian GNU/Linux 9 (stretch) on a raspberry pi 3
(Setup with these guidelines: https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian)

#### Printer model & used firmware incl. version
Anycubic Kossel, firmware 1.1.0-RC8

#### Browser and Version of Browser, Operating System running Browser
Google Chrome, Versie 62.0.3202.75 (Officiële build) (64-bits) on Debian GNU/Linux buster/sid.

#### Link to octoprint.log
2017-11-14 12:41:06,693 - octoprint.timelapse - ERROR - Could not render movie due to unknown error
Traceback (most recent call last):
  File "/home/boschie/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/timelapse.py", line 799, in _render
    p = sarge.run(command_str, stdout=sarge.Capture(), stderr=sarge.Capture())
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1411, in run
    p.run(input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1040, in run
    self.run_node(node, input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1136, in run_node
    result = getattr(self, method)(node, input, async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1282, in run_command_node
    node.cmd.run(input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 632, in run
    self.process = p = Popen(self.args, **self.kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 545, in _execute_child
    super(Popen, self)._execute_child(args, executable, preexec, *rest)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 20] Not a directory

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

It seems to me that there should be a directory. But what directory and where should it be?
Could it be there are some issues with user rights?

With kind regards,

Hendrie

kantlivelong, 2017-11-14:
In settings->OctoPrint->Folders what do you have defined for **Timelapse Folder** and **Timelapse Temp Folder**?

Do said paths exist?

boschie, 2017-11-14:
/home/boschie/.octoprint/timelapse/tmp

Yes, it does exist. It also has .jpg files in it.

kantlivelong, 2017-11-14:
You only provided one path. What is the path for each configurable option?

boschie, 2017-11-14:
[image]

kantlivelong, 2017-11-14:
Could you send fresh debug logs with the issue occuring? We are interested in the **octoprint** component. Details on how can be found [here](http://docs.octoprint.org/en/master/configuration/logging_yaml.html).

kantlivelong, 2017-11-14:
Sorry that's **octoprint.timelapse**.

boschie, 2017-11-14:
These are the last lines of the log:

```
2017-11-14 20:12:36,557 - octoprint.timelapse - DEBUG - Image /home/boschie/.octoprint/timelapse/tmp/AKM_Testing_20x20x10_20171114200221-49.jpg captured from http://127.0.0.1:8080/?action=snapshot
2017-11-14 20:12:57,597 - octoprint.timelapse - DEBUG - Capturing image to /home/boschie/.octoprint/timelapse/tmp/AKM_Testing_20x20x10_20171114200221-50.jpg
2017-11-14 20:12:57,600 - octoprint.timelapse - DEBUG - Going to capture /home/boschie/.octoprint/timelapse/tmp/AKM_Testing_20x20x10_20171114200221-50.jpg from http://127.0.0.1:8080/?action=snapshot
2017-11-14 20:12:57,761 - octoprint.timelapse - DEBUG - Image /home/boschie/.octoprint/timelapse/tmp/AKM_Testing_20x20x10_20171114200221-50.jpg captured from http://127.0.0.1:8080/?action=snapshot
2017-11-14 20:13:31,134 - octoprint.util.comm - INFO - Finished in 669.428 s.
2017-11-14 20:13:31,139 - octoprint.timelapse - DEBUG - Stopping timelapse
2017-11-14 20:13:31,143 - octoprint.timelapse - DEBUG - Waiting to process capture queue
2017-11-14 20:13:31,225 - octoprint.timelapse - DEBUG - Applying videofilter chain: [in] format=yuv420p [out]
2017-11-14 20:13:31,232 - octoprint.timelapse - DEBUG - Rendering movie to /home/boschie/.octoprint/timelapse/AKM_Testing_20x20x10_20171114200221.mpg
2017-11-14 20:13:31,262 - octoprint.timelapse - DEBUG - Executing command: /usr/bin/avconv/ -framerate 25 -loglevel error -i "/home/boschie/.octoprint/timelapse/tmp/AKM_Testing_20x20x10_20171114200221-%d.jpg" -vcodec mpeg2video -threads 1 -r 25 -y -b 5000k -f vob -vf '[in] format=yuv420p [out]' "/home/boschie/.octoprint/timelapse/AKM_Testing_20x20x10_20171114200221.mpg"
2017-11-14 20:13:31,471 - octoprint.timelapse - ERROR - Could not render movie due to unknown error
Traceback (most recent call last):
  File "/home/boschie/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/timelapse.py", line 799, in _render
    p = sarge.run(command_str, stdout=sarge.Capture(), stderr=sarge.Capture())
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1411, in run
    p.run(input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1040, in run
    self.run_node(node, input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1136, in run_node
    result = getattr(self, method)(node, input, async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 1282, in run_command_node
    node.cmd.run(input=input, async=async)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 632, in run
    self.process = p = Popen(self.args, **self.kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "build/bdist.linux-armv7l/egg/sarge/__init__.py", line 545, in _execute_child
    super(Popen, self)._execute_child(args, executable, preexec, *rest)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 20] Not a directory

```

buzztiaan, 2017-11-14:
/usr/bin/avconv/ 
remove the / behind that :D

boschie, 2017-11-14:
Serieusly.... I never thought of that... Sjeee... That did it!

Thank you so much!!!

kantlivelong, 2017-11-14:
Hooray!

closed 2017-11-14 by kantlivelong

buzztiaan, 2017-11-14:
welcome :D