OctoPrint/OctoPrint #2222: [Request] access to google drive

alasharim, 2017-11-15:
Hi, thank you for such an incredible app,

I am aware of a cloud plugin existing in octoprint called Polar Cloud which also indirectly support google drive.

the reason why I'm requesting this feature is that if I am using a slicer on a different network I could have a specific folder on my google drive as my upload folder on my octoprint which I can upload gcodes to it.

I know I can access octoprint via the internet to which I can just drag and drop but if it is possible to take out that one extra step it would be relatively more convenient.

thank you :)

foosel, 2017-11-15:
I think you might be able to do that already by mounting your google drive somewhere on your OctoPrint server and pointing OctoPrint's watched folder there. I haven't tried that though.

The tricky thing about this in any case would be authentication. Something like the regular OAuth flow with this nifty little "XYZ would like to access ABC, allow or deny?" thing won't work with OctoPrint since that requires a fixed redirect address (which we don't have with OctoPrint thanks to people being able to run it basically *anywhere* on their local network) or access to read the location string on windows opened via JS (which we don't have with OctoPrint thanks to the browser sandbox). So that makes this a case for the rather ugly "please generate an application specific password which you can find hidden 20 clicks deep in your Google account settings" approach. Kinda underwhelming :/

In any case - this is something that would only ever be implemented as a plugin, so I've marked it as such.

alasharim, 2017-11-15:
I tried mounting a google drive folder and was able to successfully mount it using "gdrivefs" and was able to list the files and folders successfully but when  tried changing my upload folder's directory to a google drive folder my whole server crashed, lol.

Its a great idea to having it as a plugin, thank you for the quick and detailed response

foosel, 2017-11-15:
I'd try changing the *watched* folder, not the upload folder. The watched folder is a special folder that will basically auto upload any files placed in it. Still, weird that stuff crashed when you changed the upload folder 🤔

ArnoutVerbeken, 2020-06-18:
I am a bit in the same situation.  I have mounted a Google Drive with _rclone_ like I do on all my other linux boxes. 
My goal was to have this mounted folder as the root of the Upload Folder, but indeed it does not work as expected.  Everything becomes very sluggish and the octoprint logfile indicates it can't write the .metadata.json file.
Next I made a softlink to my Google Drive in the default .octoprint/uploads folder, so by default everything will just be in the "local" uploadfolder and later on I would be able to move the files to their final place in Google Drive by using the FileManager plugin.
But also this does not work.  The error I get is:
`2020-06-18 19:17:32,908 - tornado.access - ERROR - 500 POST /api/files/local/CE3PRO_Cover_with_Hole.gcode (::ffff:192.168.1.189) 29.21ms
`
I do not know how the internals work, but it is the first time that an app has troubles accessing my Google Drive.

I do not know what can be done to enhance this, but it would be nice if we can use Google Drive mounts withinc OctoPrint.

djamelfel, 2020-12-10:
Hello all,
As explain by @ArnoutVerbeken I am going through the same trouble. I'm using rclone to mount my Google Drive into my system and the synchronization is working as expected. But as soon as I'm trying to sync octoprint folder to watch my Google Drive mount folder, everything is crashing.

Does someone has ever make it work ?

jneilliii, 2022-01-14:
I've had a little experience now with the GoogleDrive Backup plugin and interacting with Google's API for auth and such and might be able to get this to work similar to the way the GitFiles plugin works. I may take a stab at this over the weekend.

jneilliii, 2022-01-16:
It's in very early stages but seems to be stable enough to get some additional testers.

[image]

If you're interested in helping out and providing feedback you can find the plugin [here](https://github.com/jneilliii/OctoPrint-GoogleDriveFiles).

closed 2022-01-23 by kantlivelong