OctoPrint/OctoPrint #2223: Webcam automatically starting at boot

RodneyMcKay715, 2017-11-15:
#### What were you doing?

I reboot my Pi from time to time and i noticed that after a reboot the webcam automatically starts broadcasting a video with the mjpg streamer.
The webcamd service loads (that's fine) and then the camera automatically starts broadcasting.
Is there a way to start the service (so that i can start the broadcast with "sudo service webcamd start" over a menu item in the system commands) automatically but not start the stream yet?

#### What did you expect to happen?

That i need to start the mjpg-streamer process manually after a reboot.

#### What happened instead?

The Webcam starts streaming automatically after a reboot.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5

#### Operating System running OctoPrint

OctoPi 0.14

#### Printer model & used firmware incl. version

Anet3D A6 , Skynet 3d 2.3.1

#### Browser and Version of Browser, Operating System running Browser

Firefox Quantum 57.0

#### Link to octoprint.log

https://gist.github.com/RodneyMcKay715/17778457fa75187daaca043a8456dff9

#### Link to contents of terminal tab or serial.log

Nothing there

#### Link to contents of Javascript console in the browser

Nothing there.

I have read the FAQ.

ntoff, 2017-11-16:
I have a feeling you're confusing OctoPRINT with OctoPI.

OctoPrint is just one piece of software that runs your printer, it has no control over system services (although you can put commands in the system menu to start / stop services, it doesn't actually control the services itself but relies on the underlying OS and its commands)

OctoPi is the prebuilt raspberry pi image that contains octoprint and mjpeg streamer. 

Since your post seems to be more of a request for help, github's issue trackers are not the place for that. Please either use the mailing list https://groups.google.com/forum/#!forum/octoprint or the google+ community https://plus.google.com/communities/102771308349328485741 if assistance is directly related to octoprint / octopi but judging by the fact you have the "youtube live" plugin and you're talking about streaming, I assume you probably want to take your issues up with the author of that plugin.

foosel, 2017-11-16:
Indeed more OctoPi than OctoPrint, so closing.

As a side note - the service IS the stream.

closed 2017-11-16 by foosel