OctoPrint/OctoPrint #2226: Printer stops, OctoPrint.log shows ERROR - Exception while pushing current data

b-morgan, 2017-11-16:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
aka the Contribution Guidelines that are linked ^-- just up there
in the big yellow box.

Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ...", "I have a
question ..."), that belongs on the mailing list or the G+ community
(both linked in the "guidelines for contributing" linked above, read
them!), NOT here.

Mark requests with a "[Request]" prefix in the title please. For bug
reports fully fill out the bug reporting template (if you don't know
where to find some information - it's all described in the Contribution
Guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete ANY lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to report is still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the Contribution
Guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

1. Create dual extruder gcode file with Cura 21.08
2. Upload gcode file to OctoPrint
3. Print it

#### What did you expect to happen?

Print to finish

#### What happened instead?

Print stopped 

#### Did the same happen when running OctoPrint in safe mode?

Did not test

#### Branch & Commit or Version of OctoPrint

 OctoPrint 1.3.5 (rc/maintenance branch)

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

2017-11-16 10:41:50,461 - Recv: FIRMWARE_NAME:Marlin  FIRMWARE_VERSION:1.1.5.19 EXTRUDER_TYPE:DualExtruder v2 SOURCE_CODE_URL:https://code.alephobjects.com/diffusion/MARLIN PROTOCOL_VERSION:1.0 MACHINE_TYPE:LulzBot TAZ 6 EXTRUDER_COUNT:2 UUID:845f003c-aebd-4e53-a6b9-7d0984fde609
2017-11-16 10:41:50,463 - Recv: Cap:EEPROM:1
2017-11-16 10:41:50,464 - Recv: Cap:AUTOREPORT_TEMP:1
2017-11-16 10:41:50,466 - Recv: Cap:PROGRESS:0
2017-11-16 10:41:50,467 - Recv: Cap:PRINT_JOB:1
2017-11-16 10:41:50,468 - Recv: Cap:AUTOLEVEL:1
2017-11-16 10:41:50,469 - Recv: Cap:Z_PROBE:1
2017-11-16 10:41:50,471 - Recv: Cap:LEVELING_DATA:1
2017-11-16 10:41:50,473 - Recv: Cap:SOFTWARE_POWER:0
2017-11-16 10:41:50,474 - Recv: Cap:TOGGLE_LIGHTS:0
2017-11-16 10:41:50,476 - Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
2017-11-16 10:41:50,477 - Recv: Cap:EMERGENCY_PARSER:1

#### Browser and Version of Browser, Operating System running Browser

Firefox Quantum 57.0 (64-bit), Windows 10 Professional (64-bit)

#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/1480078/octoprint.log)

#### Link to contents of terminal tab or serial.log

[serial.log](https://github.com/foosel/OctoPrint/files/1480079/serial.log)

#### Link to contents of Javascript console in the browser

Not captured

I have read the FAQ.
Yes

foosel, 2017-11-17:
The issue appears to be that due to some transmission error the firmware received your line 32371 without a line number (even though it was sent with one): 

```
2017-11-16 12:47:22,954 - Send: N32369 G1 X133.077 Y156.018 E153.87536*109
2017-11-16 12:47:22,964 - Recv: ok N32369 P9 B3
2017-11-16 12:47:22,966 - Send: N32370 G1 X133.084 Y155.897 E153.87774*97
2017-11-16 12:47:22,979 - Recv: ok N32370 P8 B3
2017-11-16 12:47:22,981 - Send: N32371 G1 X133.131 Y155.889 E153.87867*109
2017-11-16 12:47:22,988 - Recv:  T:189.94 /190.00 B:60.00 /60.00 T0:189.94 /190.00 T1:190.03 /190.00 @:57 B@:22 @0:57 @1:45
2017-11-16 12:47:22,998 - Recv: Error:No Line Number with checksum, Last Line: 32370
2017-11-16 12:47:24,989 - Recv:  T:189.91 /190.00 B:60.01 /60.00 T0:189.91 /190.00 T1:190.00 /190.00 @:58 B@:20 @0:58 @1:46
```

And contrary to what I so far assumed, the firmware then didn't send a resend request for lastN+1 (= 32371) but instead just did nothing after complaining about the "missing" line number. So OctoPrint didn't know how to proceed and things got stuck. 

I'll look into handling this specific case better (even though I'm a bit undecided right as to what best to do here - not all lines that might be send during a print job necessarily do have a line number, it depends on configuration and also if it's even a gcode being sent). If something like this happens more often for you though you I suggest you should look into why you are getting transmission errors like this in the first place, those should rarely happen.

PS:

> **Did the same happen when running OctoPrint in safe mode?**
> Did not test

*Please always test in safe mode*. Here it wouldn't have made a difference, but you wouldn't believe how many tickets I get (or got before introducing safe mode) that were actually not an issue in OctoPrint but something caused by some third party plugin, which safe mode disables.

foosel, 2017-11-20:
Solved by the above commit. Already on `maintenance`, soon on `devel`, will be part of the 1.3.6 release.

b-morgan, 2017-11-20:
I switched the firmware from the beta 1.1.5.19 to the released 1.0.2.22 and haven't seen a repeat of this issue. I can switch back and test this fix if you give me a hint on how to set version tracking and release channel.

foosel, 2017-11-21:
My guess is that you having switched to another firmware version and not having seen this issue again is actually unrelated. The issue arises when a specific kind of mangling of the data transmitted over the line occurs, which is pretty much impossible to trigger intentionally. What I did for testing was utilize the built-in virtual printer which can trigger such error conditions, reproduced against that and then fixed it. So I fear you won't be able to actively test the fix against a real printer - chances are slim you'll actually see this error when you want to see it (which makes debugging this kind of stuff so annoying).

In any case, if you do want to test this for printing, you've got two options. You can either [switch to the "Maintenance RCs" release channel](https://github.com/foosel/OctoPrint/wiki/Using-Release-Channels) and will get an update notification once I push 1.3.6rc1 (which will include this fix) - currently I'm targeting early December for that. Or you can [manually switch to the `maintenance` branch](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image).

commit c7e248b1 (2017-11-20): Fix handling of "missing line number" errors

We so far assumed that a resend request would follow that. Alas, it
won't and we'll simply get stuck due to a missing ack from the printer.
So let's simulate that here to keep things flowing.

Fixes #2226
closed 2017-12-12 by foosel