OctoPrint/OctoPrint #2237: Issue with temperature resetting after cancelling and re-starting prints

ChrisWiltshire, 2017-11-23:
#### What were you doing?

1. Warm up both the printhead and the bed to desired temperatures while slicing.
2. Slice, with option to 'Start Printing' once slicing has finished.
3. Note the slicing has been completed, and that it's now queued for printing (pending full temperature being reached).
4. Visit the GCode viewer and have a quick look at the layering, perhaps notice an issue with the slicing settings, something you might want to change..
5. Hit cancel on the active print job.
6. Note, it is still heating as requested, the temperatures are climbing nicely.
7. Re-visit slicer, fix the issue which you spotted in the settings (ie: I wanted 100% infill for this, and not the default setting).
8. Re-slice, with Start Printing option chosen again. (It's still pre-heating, not reached temperature yet.
9. Visit GCode viewer and check it's fixed, it all looks good, so we're ready to print once that temperature has been reached.
10. The temperature has been reached, it's just started printing, ***but wait... The target temp for both the bed and the print head have fallen off, they're now unset***, but it's started printing.
11. If left like this, the print will continue through a number of layers until the filament is too cool to extrude, then the remainder of the print will be attempted with no heat in the hot-end.

#### What did you expect to happen?

I expected the GCode's temperature settings from the secondary print to be observed, and not mixed up with the print cancel request. - This is repeatable, it's happened to me twice now.

#### What happened instead?

Instead as per step 10, as soon as the secondary print request started to print, the temperature dropped off to unset, this was confirmed by the active target temperatures, the graphed temperature values including the graphed temp targets and also confirmed by the failed print where the head was moving around trying to extrude cold filament.

#### Did the same happen when running OctoPrint in safe mode?

Sorry - I have not yet had any chance to test specifically in safe mode yet. - I will add to this ticket with an attempt at safe mode if needed; I have a long print running now.

I suspect that this is further detail for an existing ticket, #2216  - (which is marked incomplete).

#### Branch & Commit or Version of OctoPrint

1.3.5 Master Branch

#### Operating System running OctoPrint

OctoPi  - 0.14.0

#### Printer model & used firmware incl. version

Anet A8 - base firmware, never upgraded, unsure of version (sorry)

#### Browser and Version of Browser, Operating System running Browser

Safari - Version 11.0.1 (13604.3.5) - Running on Mac OSX, High Sierra.

#### Link to octoprint.log

https://pastebin.com/pBxjNvdC

#### Link to contents of terminal tab or serial.log

I don't have a serial.log and the history in the terminal tab is already beyond the start of the active print. - Sorry, these can be re-captured if needed.

#### Link to contents of Javascript console in the browser

https://pastebin.com/FAAxb6eD

#### Screenshot(s)/video(s) showing the problem:

I don't have any available sorry, if needed I can re-produce further and obtain further details.

I have read the FAQ.

foosel, 2017-11-23:
> I don't have a serial.log and the history in the terminal tab is already beyond the start of the active print. - Sorry, these can be re-captured if needed.

That would be extremely helpful. Thing is, it sounds like the cancel script might be triggered (which by default disables the heaters again). But then for some reason the temperature command in your GCODE file doesn't go through again. Seeing what exactly goes over the line in that scenario would help to see what actually is going on. Without that, looking into this further is sadly just blind guess work.

ChrisWiltshire, 2017-11-23:
Understood, ok I'll aim to run a few tests and capture the serial.log for you when I next get a chance. Thank you.

ntoff, 2017-11-23:
If you can also provide a sample of your outputted gcode to compare against, and your slicer profile, I assume you're using the cura slicer plugin that ships with octoprint/octopi and not a 3rd party slicer plugin like that "full featured slicer"?

paradmat, 2017-11-23:
Hello!
Same issue for me. I think the Thermal Runaway is not working properly.
[image]

ntoff, 2017-11-24:
All your small bit of information shows is that the printer is preventing it from extruding below a certain temperature, there's no error there, that's normal behaviour.

>I think the Thermal Runaway is not working properly.

Thermal runaway protection is part of your firmware, if that really isn't functioning then you'll  have to take that up with your printer's manufacturer or firmware supplier, there's nothing OctoPrint can do there.

foosel, 2017-12-14:
@ChrisWiltshire any news on this?

closed 2018-01-22 by foosel