OctoPrint/OctoPrint #2238: Port drop only shows AUTO with no other options

Icarus1uk, 2017-11-23:
#### What were you doing?

Attempting to connect Octopi to CR-10 printer. I had this working previously and was using successfully for months, however I had to disconnect the Pi while upgrading the hotend to make space. When I reconnected it it wouldn't connect to the printer. After trying several guides online unsuccessfully I decided to reinstall Octopi.

1. Installed Octopi on Raspberry Pi Model B and booted successfully.
2. Connected Pi to Printer using USB cable. Checked USB cable by plugging into PC and was able to communicate with printer. Also checked usb port on Pi by plugging webcam in to see that it worked.
3. Set Baudrate to 115200
4. AUTO connect failed and said I needed to select port manually however there were no other options available in either the drop down list on the page or the drop down list in the settings.
5. Rebooted Pi several times and reloaded interface but no change.
6. Installed Port Listener plugin, no change.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?
Octopi to communicate with printer

#### What happened instead?
No connection and no option to select ports

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Branch & Commit or Version of OctoPrint

1.3.5

#### Operating System running OctoPrint

Octopi 0.14

#### Printer model & used firmware incl. version

Creatlity CR-10 (Hictop)

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 62.0.3202.94, Windows 10

#### Link to octoprint.log

https://1drv.ms/u/s!AvCHWOZq9WysgsdaM4sL50OWZtfItA

#### Link to contents of terminal tab or serial.log

https://1drv.ms/u/s!AvCHWOZq9Wysgsdb4YeB5V8aArwHtw

#### Link to contents of Javascript console in the browser

https://1drv.ms/u/s!AvCHWOZq9Wysgsdea91RQgee4l4HWg

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

Icarus1uk, 2017-11-23:
I tried changing the additional serial ports entry to /dev/ttyAMA0

This did then appear in the ports list but when I tried to connect I got the following in the Terminal:

Connecting to: /dev/ttyAMA0
Changing monitoring state from 'Offline' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: /dev/ttyAMA0 OSError: '[Errno 13] Permission denied: '/dev/ttyAMA0'' @ comm.py:_openSerial:1857 (hook default)

buzztiaan, 2017-11-23:
you dont have permissions

you might need to add the user thats running octoprint to the right group

`usermod -aG dailout octoprint`

or something

buzztiaan, 2017-11-23:
check what group you need with

`ls -l /dev/ttyAMA0`

Icarus1uk, 2017-11-23:
Can't access SSH right now so will give it a go later but previously this worked on USB0. Why does this no longer work?

buzztiaan, 2017-11-23:
no clue, i dont have your hardware

buzztiaan, 2017-11-23:
you could check the kernel messages when you (re)connect the usb cable to your printer with

`dmesg`

this should show the serial port being detected and the driver appointing a endpoint (like ttyUSB0 or something)

sbts, 2017-11-23:
Keep in mind it's also possible that the USB cable has failed, (or become unplugged at the other end) during your tests.

I know you said you tested on your PC and it worked, but just changing the position of the cable (especially the stresses close to the connector) may make/break the connection

foosel, 2017-11-24:
For the record: If the system OctoPrint's running on (for OctoPi that's a run off the mill Debian based Linux) is not seeing the printer as a serial device, it won't be selectable for connecting to. So the hint about checking `dmesg` is what you should do ASAP, my guess is when you plug in your printer the system doesn't even see it, or sees it under a non-standard device path that you'll need to add to your additional ports list.

Icarus1uk, 2017-11-25:
dmesg result as follow. Not overly familiar with Linux so if anyone can spot what I need please let me know.

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.28-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #998 SMP Mon May 15 16:55:39 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[    0.000000] free_area_init_node: node 0, pgdat 80c6edc0, node_mem_map b603a000
[    0.000000]   Normal zone: 1980 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b5ff4000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 223300
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=a5196906-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 871340K/901120K available (7168K kernel code, 483K rwdata, 2012K rodata, 1024K init, 778K bss, 21588K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
    lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c78f94   ( 484 kB)
       .bss : 0x80c7a000 - 0x80d3ca64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000024] Switching to timer-based delay loop, resolution 52ns
[    0.000327] Console: colour dummy device 80x30
[    0.001219] console [tty1] enabled
[    0.001267] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001338] pid_max: default: 32768 minimum: 301
[    0.001704] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001747] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002773] Disabling cpuset control group subsystem
[    0.002961] CPU: Testing write buffer coherency: ok
[    0.003029] ftrace: allocating 22386 entries in 66 pages
[    0.055943] CPU0: update cpu_capacity 1024
[    0.056010] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.056116] Setting up static identity map for 0x100000 - 0x100034
[    0.058364] CPU1: update cpu_capacity 1024
[    0.058371] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.059092] CPU2: update cpu_capacity 1024
[    0.059098] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.059762] CPU3: update cpu_capacity 1024
[    0.059769] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.059851] Brought up 4 CPUs
[    0.060025] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.060055] CPU: All CPU(s) started in HYP mode.
[    0.060081] CPU: Virtualization extensions available.
[    0.060891] devtmpfs: initialized
[    0.071937] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.072260] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.072323] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.072886] pinctrl core: initialized pinctrl subsystem
[    0.073809] NET: Registered protocol family 16
[... 179 lines cut ...]
[    3.766910] USB Video Class driver (1.1.1)
[    3.781170] systemd-journald[142]: Received request to flush runtime journal from PID 1
[    4.494396] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    4.494460] brcmfmac: power management disabled
[    4.774934] uart-pl011 3f201000.serial: no DMA platform data
[    4.892232] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:118780k SSFS
[    5.528222] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    5.528433] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.066482] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[... cut ...]
[    9.418819] Bluetooth: HCI UART driver ver 2.3
[    9.418829] Bluetooth: HCI UART protocol H4 registered
[    9.418832] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.418931] Bluetooth: HCI UART protocol Broadcom registered
[    9.569412] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.569419] Bluetooth: BNEP filters: protocol multicast
[    9.569431] Bluetooth: BNEP socket layer initialized
[   14.746643] ip_tables: (C) 2000-2006 Netfilter Core Team
[   34.144766] random: crng init done

buzztiaan, 2017-11-25:
there isn't any USB serial connected to the computer according to that dmesg

you could try removing the usb cable and replugging it , and check the bottom ~20 lines of the output afterwards?

Icarus1uk, 2017-11-25:
Tied unplugging it and plugging it into a different port get this line when I run dmesg:

[    1.467164] usb 1-1: New USB device found, idVendor=0424, idProduct=9514

Could that be it?

buzztiaan, 2017-11-25:
it should be showing usbserial as driver for it though . but that vendor and product id looks like it possible could be a serial port yeah

kantlivelong, 2017-11-25:
Looks like a USB hub.

buzztiaan, 2017-11-25:
Name: SMC9514 USB Hub

oh indeed! https://usb-ids.gowdy.us/read/UD/0424/9514

Icarus1uk, 2017-11-25:
So is that probably the USB hub built into the Pi?

buzztiaan, 2017-11-25:
yeah , maybe you're causing it to reset itself

buzztiaan, 2017-11-25:
also, that [ 1.467164] usb 1-1: New USB device found, idVendor=0424, idProduct=9514 line isnt from replugging anything

check the -BOTTOM- lines :D 

that number at the beginning is 'time in seconds since poweron'

Icarus1uk, 2017-11-25:
Here's that entire section. Can see the USB hub found bit now there's another usb device towards the bottom?

[    1.467164] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
[    1.469548] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.472719] hub 1-1:1.0: USB hub found
[    1.475117] hub 1-1:1.0: 5 ports detected
[    1.685624] systemd[1]: systemd 215 running in system mode. (+PAM +AUDIT +SELINUX +IMA +SYSVINIT +LIBCRYPTSETUP +GCRYPT +ACL +XZ -SECCOMP -APPARMOR)
[    1.690918] systemd[1]: Detected architecture 'arm'.
[    1.796899] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    1.806479] NET: Registered protocol family 10
[    1.810595] systemd[1]: Inserted module 'ipv6'
[    1.814976] systemd[1]: Set hostname to <octopi>.
[    1.817761] systemd[1]: Initializing machine ID from random generator.
[    1.820593] systemd[1]: Installed transient /etc/machine-id file.
[    1.927214] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[    1.929876] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber

buzztiaan, 2017-11-25:
the -BOTTOM- , use this ;

`dmesg | tail -n 20`

Icarus1uk, 2017-11-25:
[    4.745875] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:118780k SSFS
[    5.279189] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    5.279497] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.785295] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    6.786070] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    9.236388] Bluetooth: Core ver 2.22
[    9.236471] NET: Registered protocol family 31
[    9.236476] Bluetooth: HCI device and connection manager initialized
[    9.237646] Bluetooth: HCI socket layer initialized
[    9.237664] Bluetooth: L2CAP socket layer initialized
[    9.237704] Bluetooth: SCO socket layer initialized
[    9.244096] Bluetooth: HCI UART driver ver 2.3
[    9.244104] Bluetooth: HCI UART protocol H4 registered
[    9.244107] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.244227] Bluetooth: HCI UART protocol Broadcom registered
[    9.398110] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.398117] Bluetooth: BNEP filters: protocol multicast
[    9.398128] Bluetooth: BNEP socket layer initialized
[   12.588445] ip_tables: (C) 2000-2006 Netfilter Core Team
[   66.520692] random: crng init done

buzztiaan, 2017-11-25:
yep, so nothing happened, do you get extra lines when you plug in a known working thing? like a usb flashdrive?

Icarus1uk, 2017-11-25:
Yeah, when I plug in webcam that seems to register fine

buzztiaan, 2017-11-25:
ok, so either your electronics are broken , or the usb cable , or its usb interface

kantlivelong, 2017-11-25:
Can you provide the result of running `lsusb` both prior and post plugging the printer in? Please also make sure that the camera is unplugged and the printer is already on to avoid powering the controller from the Pi.

Icarus1uk, 2017-11-25:
Before and after are identical so clearly not picking anything up. But still registers as a device when I plug into the PC

Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

buzztiaan, 2017-11-25:
yeah, not sure man, maybe ask the CR-10 people why it stopped working on USB?

kantlivelong, 2017-11-25:
Being that it seems as if it could be resetting the hub due to power draw my only other suggestion would be to try something like a USB condom to prevent power draw entirely.

kantlivelong, 2017-11-25:
Going to close this though as it isn't related to OctoPrint. Feel free to join us on IRC.

closed 2017-11-25 by kantlivelong