OctoPrint/OctoPrint #2240: Uploading files via curl with path command duplicates file name

calonmerc, 2017-11-23:
#### What were you doing?

Trying to upload files to Octoprint using curl on Windows.

Run the following command:
`curl -k -H "X-Api-Key: <redacted>" -F "select=true" -F "print=false" -F "file=@C:\Users\Kyle\3D Printer\gcode\filename.gcode" -F "path=MPSM/filename.gcode" "http://<redacted>/api/files/local"`

As well as:
`curl -k -H "X-Api-Key: <redacted>" -F "select=true" -F "print=false" -F "file=@C:\Users\Kyle\3D Printer\gcode\filename.gcode" -F "path=MPSM/filename.gcode" "http://<redacted>/api/files/local"`
`curl -k -H "X-Api-Key: <redacted>" -F "select=true" -F "print=false" -F "file=@C:\Users\Kyle\3D Printer\gcode\filename.gcode" -F "path=MPSM/filename" "http://<redacted>/api/files/local"`
`curl -k -H "X-Api-Key: <redacted>" -F "select=true" -F "print=false" -F "file=@C:\Users\Kyle\3D Printer\gcode\filename.gcode" -F "path=MPSM/" "http://<redacted>/api/files/local"`
`curl -k -H "X-Api-Key: <redacted>" -F "select=true" -F "print=false" -F "file=@C:\Users\Kyle\3D Printer\gcode\filename.gcode" -F "path=MPSM/*" "http://<redacted>/api/files/local"`
`curl -k -H "X-Api-Key: <redacted>" -F "select=true" -F "print=false" -F "file=@C:\Users\Kyle\3D Printer\gcode\filename.gcode" -F "path=MPSM" "http://<redacted>/api/files/local"`

#### What did you expect to happen?

I expected the file to be uploaded to Octoprint in the folder MPSM/

#### What happened instead?

The file was uploaded, but with an additional folder. `MPSM/filename.gcode/filename.gcode`

#### Did the same happen when running OctoPrint in safe mode?

Same problem occured in safe mode.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5 (master branch)

#### Operating System running OctoPrint

OctoPi on Raspberry Pi 3

#### Browser and Version of Browser, Operating System running Browser

Using curl on Windows

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

https://pastebin.com/zqwX6FmX

I have read the FAQ.

buzztiaan, 2017-11-23:
`curl -k -H "X-Api-Key: <redacted>" -F "select=true" -F "print=false" -F "file=@C:\Users\Kyle\3D Printer\gcode\filename.gcode" -F "path=MPSM/filename.gcode" "http://<redacted>/api/files/local"`

> The file was uploaded, but with an additional folder. MPSM/filename.gcode/filename.gcode

isn't that matching 100% what your command says?

foosel, 2017-11-23:
@calonmerc forgot to include the [earlier thread on G+ on this](https://plus.google.com/u/0/+KyleMenigoz/posts/hDa9XBzzuY9?cfem=1), hence maybe a bit of confusion here.

The issue is that

> ``curl -k -H "X-Api-Key: <redacted>" -F "select=true" -F "print=false" -F "file=@C:\Users\Kyle\3D Printer\gcode\filename.gcode" -F "path=MPSM" "http://<redacted>/api/files/local"``

is also apparently duplicating the file name. That *should* lead to `MPSM/filename.gcode` instead of `MPSM/filename.gcode/filename.gcode` but apparently it doesn't.

ntoff, 2017-11-23:
`curl -k -H "X-Api-Key: <redacted>" -F "select=ftrue" -F "print=false" -F "file=@success.gcode" -F "path=MPSM/" "http://woody.local/api/files/local"`

This works for me using linux mint's curl to send the file, it doesn't duplicate anything. Maybe it's an issue with the windows version of curl?

I get this as the result:

    {
      "done": true, 
      "files": {
        "local": {
          "name": "success.gcode", 
          "origin": "local", 
          "path": "MPSM/success.gcode", 
          "refs": {
            "download": "http://woody.local/downloads/files/local/MPSM/success.gcode", 
            "resource": "http://woody.local/api/files/local/MPSM/success.gcode"
          }
        }
      }

calonmerc, 2017-11-24:
I just did an update of curl... I should have checked what version it was, but now it's working.

foosel, 2017-11-24:
Ah, so a curl problem after all. Closing this then :)

closed 2017-11-24 by foosel