OctoPrint/OctoPrint #2241: New file type?

cpdevelops, 2017-11-23:
Is there any chance that support (or a plugin adding support) for the BFB 3D Touch printer? At the moment, you are required to use outdated software which creates GCode (but also seems to run through some type of postprocessor, not sure what it modifies though). I would like to be able to connect it to an octoprint instance without making crazy modifications (I.e replacing the controller board so that it uses pure gcode). 

If this is not possible for someone else to write a plugin or add support., it would be much appreciated if you could point me to the proper documentation.

Thanks.

cpdevelops, 2017-11-23:
Fixed!

foosel, 2017-11-24:
I'd start with [the docs on the "octoprint.filemanager.extension_tree" hook](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-filemanager-extension-tree). But it's not clear to me what further steps might be necessary.

What format are those files? What protocol does the printer speak over the serial line? Does it even have a serial connection or how does that work? Questions after questions :)

cpdevelops, 2017-11-29:
Sorry for the late response.

> What format are those files?
The files seem to be gcode with some unknown post processing afterwards. Slicing and processing the files would not be needed for my purposes, I only need to be able to upload,queue, and send files to the printer. 

> What protocol does the printer speak over the serial line? Does it even have a serial connection or how does that work?
It has a USB-B connection which allows it to connect tethered to the printer and has one USB slot for giving print files untethered. It does not seem to have a serial connection, but maybe there could be some connection straight to the controller board?

foosel, 2017-11-30:
In that case just adding a new file type won't help you. If it doesn't have a serial connection out of the box you are either looking at building your own serial factory to emulate one (basically a small step further than what the GPX plugin does) or at outright replacing the whole comm layer. Neither is going to be a case of "add a couple of lines and be happy" I fear.

closed 2017-12-28 by cpdevelops