OctoPrint/OctoPrint #2243: GCode Viewer: confused when parts are printed once at a time

bzed, 2017-11-25:
#### What were you doing?

I'm printing some parts with the 'one at a time' setting of cura, so as long as there is enough space to move the head, it will print part after part to reduce movements.

#### What did you expect to happen?

I've expected the gcode viewer to show me how each part is being printer and in which order.

#### What happened instead?

Instead it showed a mix of all parts. Tracking whats being printed at the moment seemed to be correct

#### Branch & Commit or Version of OctoPrint

1.3.5

#### Operating System running OctoPrint

Debian Stretch

#### Printer model & used firmware incl. version

cr-10 s4 marlin 1.1.6

#### Browser and Version of Browser, Operating System running Browser

google chrome 63.0.3239.9-1

#### Link to octoprint.log
http://paste.debian.net/997554/

#### Link to contents of Javascript console in the browser
http://paste.debian.net/997555/

#### Screenshot(s)/video(s) showing the problem:

I'll follow up with screenshots and a gcoe example as soon as my own server is up and running again. Jus twant to document it before I forget about it.

ziceva, 2017-11-25:
I can confirm this ... I noticed this behavior too ...
The g-code viewer shows the the current layer (including the estimates of time and filament) the same way as if it was a "all  at once" ... when the next object is printing, it shows the same thing again for each layers ...

bzed, 2017-11-25:
So here are my screenshots:

First layer at print start:
http://shell.bzed.at/~bzed/.publish/2017-11-25-9xb9RNnS9Z0/Screenshot_20171125_153322.png

Tracking current layer - the upper right part is being printed at the moment, the other parts are done already.
http://shell.bzed.at/~bzed/.publish/2017-11-25-9xb9RNnS9Z0/Screenshot_20171125_173751.png

"Next Layer", starting from the current one
http://shell.bzed.at/~bzed/.publish/2017-11-25-9xb9RNnS9Z0/Screenshot_20171125_173816.png (WTF?)

gcode;
http://shell.bzed.at/~bzed/.publish/2017-11-25-Qa0M1dtRPJE/CCSA_Control_Top.gcode

foosel, 2017-11-27:
And now @bzed's even opening tickets! :P Hello buddy :D

That issue is caused by the included GCODE viewer modelling the GCODE file's data by layer (essentially z-height -> list of commands). As part of implementing something requested in #1973 I also [looked into changing the whole data model](https://gist.github.com/foosel/a61f4a710d339d02dbe140e11b2b86bc) to better detect consecutive layers and handle them regardless of layer Z, but that produced some horrible side effects with some of my test files that then started to go all over the place. The problem here is reliably detecting starts of new layers without producing false positives for stuff like z-hop or nozzle cleaning routines and also ideally handling vase mode gracefully (which right now it doesn't at all), and doing that on the go while reading and parsing the file.

Nothing trivial sadly and I currently find it very hard to justify the work it would necessitate to get done correctly.

bzed, 2017-11-27:
@foosel of course :-P I'd prefer to fix them, but right now I can't find the time. Btw, even my copy&paste foo took some time to make @GitIssueBot happy, would be easier if it would tell whats missing ;)

regarding the issue: maybe it makes sense to split the new-layer detection and new-part/single-part mode detection into two different issues. If the printe rmoves to a new area and down to the height of layer 0, then it is a new part and layer/print handling could start from there. Somehow that cries for a FSM ;)

foosel, 2017-11-28:
"The height of layer 0" is not necessarily that easy to determine. Add some pre print purging routines to the start gcode, and all of a sudden it's tricky to detect what even is purging and what actually printed object, let alone what's the first layer. The current implementation tries to circumvent this by assuming that purging happens out of bounds of the actual print bed, but that isn't necessarily the case and there are situations where the viewer also shows a weird view in such cases.

All in all, the problem here is the z-height based model approach that was inherited from the origin of the gcode viewer. Works great for run of the mill straight up prints, but runs into some serious limitations when confronted with things like consecutive models on a plate, purging routines, z-hops, vase mode, ... Your observed issue is sadly merely one expression of that general problem. And solving that like you suggest would already be a case of "ok, let's see what that now broke with all the other exceptions" since it would necessitate exactly what I already tried - getting away from an enforced "what's on the same z is the same layer" and instead detecting z switches to create a new layer. And *that's* what's causing the major headache here since it needs to take care of all the aforementioned cases, or it will work flawlessly for your specific scenario but break left and right for a large group of the user base at the same time.

---

Re the bot, you can't claim you weren't warned :P

Pre-filled ticket text:

> When reporting a bug do NOT delete ANY lines from the template but those enclosed in [ and ]

Contribution guidelines:

> Copy-paste this template **completely** (or use the version that gets pre-filled into the "new issue" form). Do not skip any lines or the bot will complain!

foosel, 2022-02-25:
This got (accidentally 😅) fixed in #4430, which will be part of 1.8.0

closed 2022-05-17 by cp2004