OctoPrint/OctoPrint #2244: When installing No-Ip it disables internet connection

Jandro5vq, 2017-11-25:
# When installing No-IP it disables internet connection

Hi, thank you for reading this.

I have an **Octopritn** installation in a **RPI 3 B** with an **Anet A6** printer an it works very well, but I installed in the **Octoprint no-ip** to access the printer from the internet, but a day after the installation the router _disables the internet connection_ when **RPI** is on I mean when **RPI** is on I can't browse the internet whit any device.

**_Can someone help me?_**

Thank you again

ntoff, 2017-11-26:
Going to need a LOT more info there champ, but I suspect you're in completely the wrong place anyway. OctoPrint has no control over your internet connection, that's handled by the underlying operating system.

If you're talking about installing a plugin, and that plugin breaks your internet, then you'll have to take that up with the author of that plugin.

Jandro5vq, 2017-11-26:
Sorry, yes, I notice that I was in the wrong place an I erase it thinking it would disappear but it continues here thank you any ways.

Sorry for being a nuisance

ntoff, 2017-11-26:
You can click the "close issue" button which should be next to the comment button for you, github isn't like a forum, you don't just delete things, it's more like the NSA, you just hide (close) things and hope no one finds it. -_-

closed 2017-11-26 by Jandro5vq