OctoPrint/OctoPrint #2245: jogging axes doesn't trigger firmware endstops

qqazxswedc, 2017-11-26:
#### What were you doing?

jogging the x axis (The problem happens with the y axis too, i just noticed it first with the x.)

Ideally provide exact steps to follow in order to reproduce your problem:

1. jog the x or y axis <i>away</i> from the location of your <b>hardware</b> endstop. 
2. continue jogging it all the way, until it cannot physically move any more

#### What did you expect to happen?

I expected that the (marlin)firmware endstops that are set on my machine to trigger

#### What happened instead?

octoprint ignored the fact that a firmware endstop was triggered

#### Did the same happen when running OctoPrint in safe mode?

yes

#### Branch & Commit or Version of OctoPrint

latest version as of 11/26/17

#### Operating System running OctoPrint
linux

#### Printer model & used firmware incl. version

printrbot simple metal, marlin firmware

#### Browser and Version of Browser, Operating System running Browser

firefox 57/firefox quantum tunning on ubuntu 16.04

I have read the FAQ.

buzztiaan, 2017-11-26:
did you home the axis prior to this?

ntoff, 2017-11-26:
OctoPrint doesn't care what your hardware end stops are doing, this is a firmware issue if your printer doesn't stop when it hits an endstop. Configure your firmware to obey end stops always (as opposed to only when  homing), and set up your build volume properly inside your firmware. All OctoPrint does is send your printer commands, it's up to the firmware to then execute (or not) that command.

foosel, 2017-11-27:
See also [the FAQ on this](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-is-telling-my-printer-to-do-wrong-things).

closed 2017-11-27 by foosel