OctoPrint/OctoPrint #2247: Hardware detection fails on a VM install

ntoff, 2017-11-26:
#### What were you doing?

Attempting to use the maintenance / dev branch

1. git checkout devel
2. python setup.py clean && python setup.py install && octoprint serve

#### What did you expect to happen?

Octoprint to start

#### What happened instead?

hardware detection fails because the cpu frequency is detected as "None"

    (oprint) mint@mint-VirtualBox ~/octoprint $ python
    Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
    [GCC 5.4.0 20160609] on linux2
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import psutil
    >>> print(psutil.cpu_freq())
    None

#### Did the same happen when running OctoPrint in safe mode?

yes, the issue is with a bundled package, not a 3rd party plugin

#### Branch & Commit or Version of OctoPrint

>Can be found in the lower left corner **of the web interface**. ALWAYS INCLUDE.

-_- not sure if mocking me? or...

OctoPrint 1.4.0.dev673+g312760f (devel branch)

#### Operating System running OctoPrint

mint 18 in a virtualbox virtual machine

#### Printer model & used firmware incl. version

n/a

#### Browser and Version of Browser, Operating System running Browser

n/a

#### Link to octoprint.log

https://gist.github.com/ntoff/bfbea594b1cb3d2e963a0af91c8bc977

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

n/a

#### Screenshot(s)/video(s) showing the problem:

n/a

I have read the FAQ.

foosel, 2017-11-27:
This should already be fixed as of 26789a2bb7c6fae8d756b4ca8236d1cfb5491be1 which was merged to `devel` as well in ad246deb9e7f587b75ce1a0eb6ba6f52900c2029

Can you update and verify?

ntoff, 2017-11-27:
Weird, I tried it last night / this morning and nothing, but yeah it's working now. I must have missed the "git pull" after checking out devel. Sorry :(

foosel, 2017-11-27:
Happens ;)

closed 2017-11-27 by foosel