OctoPrint/OctoPrint #2251: [Request] [Feature] Get filename of selected file from the API

Nedevski, 2017-11-28:
I am creating an Alexa skill and I've found that I have no way to get information about the currently selected file via the API.

I need only the filename, but if it's the full file object, that would be useful too.

foosel, 2017-11-28:
Uhm... http://docs.octoprint.org/en/master/api/job.html#retrieve-information-about-the-current-job

Nedevski, 2017-11-28:
Oh, so when you select a file, it becomes visible in the job, even if it's not started yet?

If that's so - my bad.

foosel, 2017-11-28:
Yes. A job is a job even before it's started. Current modelling only ever has one active job, and that maps 1:1 to the currently selected file in the UI. I'll add a note to the docs though, you are the second person in the past two months that's stumbled over this :)

May I ask where you were looking for a way to retrieve the currently selected file? In the [file operations section](http://docs.octoprint.org/en/master/api/files.html)?

Nedevski, 2017-11-28:
No, I was looking in the job operations section, I just assumed that a job exists only when the printer is printing.

But I admit that I completely overlooked the Retrieve info about current job endpoint, since I assumed it's of no use to me. My bad there.

Thanks for the fast responses!

closed 2017-11-28 by Nedevski