OctoPrint/OctoPrint #2253: Web cam configure

userosos, 2017-11-29:
I want start work on octopi on an raspberry with an web cam. I know command for start and work with the camera.

The command is ./mjpg_streamer -i "input_uvc.so -d /dev/video0 --fps 30 --led off --quality 50" -o "output_http.so -p 8080"

The webcam is sven IC-310. 
I don't know where i need add the line for that my webcam will start automatic with run the octopi server.
If i add the line in /home/pi/mjpg-streamer/start.sh it not work.
I can kill default daemon and after it start my script but it is bad.

Will be good if i can edit command for start my webcam inside webshell.

ntoff, 2017-11-29:
Please use the mailing list or google+ community (linked to by the bot) for support requests, and see the FAQ on why OctoPrint has no webcam controls https://github.com/foosel/OctoPrint/wiki/FAQ#my-webcam-is-not-working.

Also your question on where to put your command line is answered in the FAQ https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-change-mjpg-streamer-parameters-on-octopi

kantlivelong, 2017-11-30:
I'd also like to direct you to the [OctoPi repo](https://github.com/guysoft/OctoPi).

closed 2017-11-30 by foosel

userosos, 2017-11-30:
I add the line:
camera_usb_options="-i "input_uvc.so -d /dev/video0 --fps 30 --led off --quality 50" -o "output_http.so -p 8080""

in /boot/octopi.txt
And it not work. I understand that the problem directly not related with octoprint. But inside OctoPi distribution is built-in webcam service and i can't changes some parameters the service.

userosos, 2017-11-30:
I add the line in /boot/octopi.txt and it work!
camera_usb_options="-r 640x480 -f 30 -q 80"

RoqueBelda, 2017-12-15:
Hello, I have a similar problem...

I have install Octoprint in my Orange Pi Lite over the official distribution ARMbian for this model. Later i have install OctoPrint from this repository and It is work very well.

But, now i want to configure my Cam, and i don't found octopi.txt in my folders... Some idea?

Another question...  My Cam is  "Microdia Win2 PC Camera" according with the " lsusb" command. It is compatible? In the table i found something similar: "LogiLink WebCam (Microdia Sonix USB 2.0 Camera)" The Cam is a Ixium

Thanks

foosel, 2017-12-15:
For questions or other kind of help or support, please get in touch on the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

userosos, 2018-03-20:
Need install https://github.com/jacksonliam/mjpg-streamer external. After it need edit:
sudo nano /root/bin/webcamd
~53 line need "./mjpg_streamer" to "mjpg_streamer"
" echo Running mjpg_streamer -o "output_http.so -w $camera_http_webroot $camera_http_options" -i "$input"
        LD_LIBRARY_PATH=. mjpg_streamer -o "output_http.so -w $camera_http_webroot $camera_http_options" -i "$input" &"