OctoPrint/OctoPrint #2254: [Bug?] With the newest firmware update Lulzbot Taz6

keawepoo, 2017-11-30:
Hello, after upgrading my Lulzbot Taz6 firmware my print would stop when switching to T1 or really after a min or so of printing. I came to this from taking OctoPrint offline and using the SD and USB and they both worked fine, only the OctoPrint would freeze up. (the temp went down from 240 to 175 and never went back up it just stayed at 175 for almost 10mins until I killed the print)
video: https://photos.app.goo.gl/NblqrYdrMKAQKaZF3

Thank you. Please let me know if you need more info...

foosel, 2017-11-30:
Please provide _all_ requested information. You are missing the in this case most crucial bit - a `serial.log` or at the very least the contents of your Terminal tab plus the contents of your JS console. Without that information this ticket cannot be looked into and will be closed. Thanks.

> after upgrading my Lulzbot Taz6 firmware my print would stop when switching to T1 or really after a min or so of printing

Without being able to see the communication between OctoPrint and your printer, this makes it sound like a firmware bug introduced with the firmware upgrade, probably only reproducible when talking to the printer via serial (what OctoPrint does). But at this point that's just blind guessing.

keawepoo, 2017-11-30:
Terminal: https://gist.github.com/keawepoo/b43b59f244fe2668def7b1310040b0bc
serial.log, its blank... 
https://drive.google.com/open?id=1tH1gLOp57FNvudJw69MrvvdIr40Po5VL

foosel, 2017-11-30:
> serial.log, its blank...

Yes, you have to enable it :) It's disabled by default due to performance reasons. See https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#where-can-i-find-those-log-files-you-keep-talking-about

keawepoo, 2017-11-30:
serial.log: https://gist.github.com/keawepoo/cad1b85e996b703f5afb1445560d5d15

foosel, 2017-12-01:
My guess is the problem is this:

```
2017-11-30 22:51:55,085 - Send: N349 G1 X101.474 Y147.171 E4.7342*88
2017-11-30 22:51:55,102 - Recv: Error:No Line Number with checksum, Last Line: 348
```

And that is the exact same failure pattern as reported in #2226 (also with a TAZ6 I might add), making this a duplicate. The OP of that ticket [reported](https://github.com/foosel/OctoPrint/issues/2226#issuecomment-345782638) that switching from "beta 1.1.5.19 to the released 1.0.2.22" firmware made the issue disappear for them. In any case though there was a problem still with OctoPrint not handling that particular `No Line Number with checksum` error properly, causing communication to run into issues. That has been solved though and a fix for that will be pushed with 1.3.6.

To quote myself from the other ticket:

> If something like this happens more often for you though you I suggest you should look into why you are getting transmission errors like this in the first place, those should rarely happen. 

I also spotted another communication error in your terminal output:

```
Send: N238 G1 X232.105 Y12.002 E6.6366*110
Recv:  T:220.13 /220.00 B:116.80 /120.00 T0:218.33 /220.00 T1:220.13 /220.00 @:104 B@:103 @0:127 @1:10N2ok N238 P1 B3
```

Note how the `ok N238 P1 B3` there is attached to the temperature output line instead of on its own line like it should be. Looks like either a firmware bug (unlikely though, because then I'd expect it to happen more often) or something in your printer comms getting confused (that could also be due to a firmware bug, I've seen a lot of interesting stuff happen on the serial line when interrupt routines of the firmware take too long to complete, causing loads of communication errors and corrupt lines).

See [this FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-keeps-running-into-communication-errors-and-timeouts) for common causes for this other than firmware.

foosel, 2017-12-01:
Closing because it looks very much like a duplicate and/or communication and/or firmware issues. Comments are still open though.

closed 2017-12-01 by foosel