OctoPrint/OctoPrint #2256: [RC Feedback] Feedback on the 1.3.5rc1 Release Candidate

foosel, 2017-12-01:
Please provide *general* feedback on your experience with the 1.3.6rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

Kunsi, 2017-12-01:
Copy-Pasting IRC logs:

16:05:46 <Kunsi> foosel: the firstrun assistant asks me to enable a plugin blacklist … would be nice to include a link to that - or some other way to view it
16:06:25 <+foosel> Kunsi, good point.
16:07:01 <+foosel> Kunsi, just so you know for now: http://plugins.octoprint.org/blacklist.json
16:07:51 <Kunsi> thanks
16:08:20 <Kunsi> even better: match the plugins agains currently installed versions, and show if currently installed plugins would be affected

foosel, 2017-12-01:
@Kunsi that's now changed by the above commit:

[image]

[image]

See also the newly added page http://plugins.octoprint.org/blacklist/ for the mentioned "more human readable" format.

And minimally invasive enough that it will definitely make it into the final 1.3.6.

schorsch3000, 2017-12-01:
I just did a few small prints so far (mainly to dail my new cam in), everything seems fine so far, exept the 

`Navbar Temperature Plugin (0.8)` which basically varnished.
... I thing is should report this to the plugin, right?

ntoff, 2017-12-01:
Shouldn't that plugin blacklist appear in the plugin manager page (rather than server settings)?

ripp2003, 2017-12-02:
Same problem like schorsch3000, upgrade was successful, but Navbar Temp plugin did not work.... 
Thanks for this RC ;-)

ntoff, 2017-12-02:
Here's why the navbar temp plugin doesn't work https://github.com/foosel/OctoPrint/issues/2246 and https://github.com/foosel/OctoPrint/issues/2200 so yes, you'll need to report the issue to the plugin author so they can update the plugin to stop using global variables. Here's the octoprint blog post about it http://octoprint.org/blog/2017/12/01/heads-up-plugin-authors/

JohnOCFII, 2017-12-02:
I've uploaded and printed twice without issue.

I'm using the OctoSlack and Print History Plugins, and they seem fine.

I've generated and viewed a timelapse successfully.

malnvenshorn, 2017-12-02:
I encountered one issue. Setting a temperature offset to one tool resets a previously set offset on other tools. But it appears to be only the frontend showing wrong values.

ctgreybeard, 2017-12-03:
I have printed a couple of things since upgrading then tonight I had a fatal checksum error. The print stopped cold with a communications error. octoprint.log is attached which shows the problem.

One thing I find disturbing is that I wasn't notified by the Telegram plug-in. I'm not sure this is an issue with OctoPrint or with the plug-in but I think I should have gotten some indication of the failure.

I won't try this again until tomorrow. I can provide the gcode if you need it.

[ctgreybeard.log.tar.gz](https://github.com/foosel/OctoPrint/files/1524390/ctgreybeard.log.tar.gz)

cosmith71, 2017-12-03:
I had a similar checksum error twice.  It seemed to stop in the same spot, even after re-slicing and re-sending.

[new 1.txt](https://github.com/foosel/OctoPrint/files/1524407/new.1.txt)

ctgreybeard, 2017-12-03:
I'm running the same gcode through the virtual printer. It's taking a while but I noticed some artifacts in the virtual serial log. A small sample looks like this:

```
2017-12-02 22:48:48,387 >>> ok
2017-12-02 22:48:48,391 <<< N189466 G1 X73.728 Y88.187 E0.1508*109
2017-12-02 22:48:48,458 <<< E0.1508*109
2017-12-02 22:48:48,460 <<< N189467 G1 X77.058 Y88.187 E0.2030*101
2017-12-02 22:48:48,461 >>> ok
2017-12-02 22:48:48,468 <<< 7 E0.2030*101
2017-12-02 22:48:48,470 <<< N189468 G1 X78.295 Y86.045 E0.1508*106
2017-12-02 22:48:48,471 >>> ok
2017-12-02 22:48:48,487 <<< 5 E0.1508*106
2017-12-02 22:48:48,489 <<< N189469 G1 X81.625 Y86.045 E0.2030*111
2017-12-02 22:48:48,490 >>> ok
2017-12-02 22:48:48,506 <<< 5 E0.2030*111
2017-12-02 22:48:48,508 >>> ok
2017-12-02 22:48:48,509 <<< N189470 G1 X82.862 Y88.187 E0.1508*101
2017-12-02 22:48:48,543 >>> ok
2017-12-02 22:48:48,543 <<< 7 E0.1508*101
2017-12-02 22:48:48,547 <<< N189471 G1 X86.193 Y88.187 E0.2030*106
2017-12-02 22:48:48,562 >>> ok
2017-12-02 22:48:48,564 <<< 7 E0.2030*106
2017-12-02 22:48:48,566 <<< N189472 G1 X87.429 Y86.045 E0.1508*96
2017-12-02 22:48:48,629 >>> ok
2017-12-02 22:48:48,631 <<< 5 E0.1508*96
2017-12-02 22:48:48,634 <<< N189473 G1 X90.760 Y86.045 E0.2030*100
2017-12-02 22:48:48,665 >>> ok
2017-12-02 22:48:48,667 <<< E0.2030*100
2017-12-02 22:48:48,669 <<< N189474 G1 X91.997 Y88.187 E0.1508*104
2017-12-02 22:48:48,731 >>> ok
2017-12-02 22:48:48,733 <<< 7 E0.1508*104
2017-12-02 22:48:48,736 >>> ok
2017-12-02 22:48:48,737 <<< N189475 G1 X95.327 Y88.187 E0.2030*97
2017-12-02 22:48:48,741 <<< N189476 G1 X96.486 Y86.180 E0.1413*98
2017-12-02 22:48:48,804 >>> ok
2017-12-02 22:48:48,806 <<< E0.1413*98
2017-12-02 22:48:48,808 <<< N189477 G1 X97.212 Y87.235 E0.0781*108
2017-12-02 22:48:48,839 >>> ok
2017-12-02 22:48:48,841 <<< E0.0781*108
2017-12-02 22:48:48,844 <<< N189478 G1 X98.978 Y89.846 E0.1921*110
2017-12-02 22:48:48,906 >>> ok
2017-12-02 22:48:48,908 <<< 6 E0.1921*110
2017-12-02 22:48:48,910 <<< N189479 G1 X99.615 Y90.824 E0.0711*106
2017-12-02 22:48:48,915 <<< 4 E0.0711*106
2017-12-02 22:48:48,916 >>> ok
2017-12-02 22:48:48,918 <<< N189480 G1 X96.564 Y90.824 E0.1859*100
```

Notice that the endings of many lines are repeated. I can't say if this is a regression because I've never looked at it before.

ctgreybeard, 2017-12-03:
Well, doesn't look like that is going to trigger the problem, the line numbers don't match for the same gcode line and so the checksum is different:

```
╰─➤  grep 251731 logs/octoprint.log
2017-12-02 17:28:30,879 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: checksum mismatch, Last Line: 251731
| Send: N251731 G1 X111.123 Y91.986 E0.0018*86
| Recv: Error:checksum mismatch, Last Line: 251731
| Send: N251731 G1 X111.123 Y91.986 E0.0018*86
| Recv: Error:checksum mismatch, Last Line: 251731
| Changing monitoring state from 'Printing' to 'Error: checksum mismatch, Last Line: 251731
╭─pi@camera1 ~/.octoprint
╰─➤  grep X111.123 logs/plugin_virtual_printer_serial.log
2017-12-02 21:54:56,910 <<< N62887 G1 X111.123 Y115.330 E0.1881*85
2017-12-02 23:13:57,686 <<< N250093 G1 X111.123 Y91.986 E0.0018*88
```

ntoff, 2017-12-03:
As stated in the original post, if you have actual bugs that break things, please file a standalone report following the issue template (the one you should see when you open a new issue ticket).

goeland86, 2017-12-03:
I had a problem during the upgrade - apparently some of the plugins installed weren't blacklisted but still caused issues. The console showed OctoPrint started but it wasn't loading in any browser (FF, Chrome or Edge). I had a mention about permissions on a .git folder so I tried to change that - as we maybe muck around with perms a bit on Umikaze's install of OctoPrint.

Even getting rid of all warnings I had to re-create my ~/.octoprint folder from scratch to get it to start... I had the NavBar temp, but also a few others which I don't remember installed. That said, everything's working now, except for blacklisted plugins (d'oh!)

Crowlord, 2017-12-03:
General feedback as I stupidly put it on a Machine I have orders to print on so really cannot troubleshoot. 

Received the checksum error as above on my Prusa Mk2 FW is latest. I have had to roll back to 1.3.5 and am currently falling back to SD card. 

As there are a large number of Mk2s out there are in the wild this could be a big breaking bug. Sorry, I cant be more helpful though. I shouldn't really even have been on the RC!

ctgreybeard, 2017-12-03:
Hmmm, I am also on a Prusa MK2 with 3.1.0 firmware. I installed that because of the upgrade to OctoPrint compatibility (busy replies changing filament specifically)

JohnOCFII, 2017-12-03:
As a data point, my successful OctoPrint RC testing was on a MK2, but with the older 3.0.12 firmware (mostly 3.0.12 -- but with the Linear Advance feature added).

chippypilot, 2017-12-03:
I've run a few small prints with success. I like the improved notification popups.

Pi Zero W and camera
Anet A6

Crowlord, 2017-12-03:
I guess I should elaborate these were long prints and it took several hours to exhibit. (33Mb gcodes)

fiveangle, 2017-12-03:
Looks like Marlin auto-report temperature messages are no longer being suppressed:

> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.09 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0
> Recv:  T:21.25 /0.00 B:23.12 /0.00 @:0 B@:0

cosmith71, 2017-12-03:
> I guess I should elaborate these were long prints and it took several hours to exhibit. (33Mb gcodes)

Mine were as well.  The checksum error occurred approximately 5 hours in, at the same spot both times.  I'm also using an Original Prusa MK2 with 3.1.0 firmware.  I rolled back to Octoprint 1.3.5 and am now 12 hours in and well past the point 1.3.6rc1 died.  

I'll look at putting in a proper bug report, but my printer is tied up and I don't have all the information for a complete report.  It may take me a while.  Maybe someone will beat me to it.

JohnOCFII, 2017-12-04:
I just had the checksum error as well.  My previous (successful) prints were all under two hours.  This print failed just after the two hour mark.

```| Recv: Error:checksum mismatch, Last Line: 118727
2017-12-03 16:31:36,764 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 118728, current line = 118730```

I'll do a proper issue entry after dinner

ctgreybeard, 2017-12-04:
Reverting to 1.3.5 let the previous failed print complete properly.

As a test I checked the checksum on the line that failed and it was correct.

JohnOCFII, 2017-12-04:
OK, I've created a separate Issue for the checksum error: https://github.com/foosel/OctoPrint/issues/2262

Unfortunately, I don't have time to re-test that print, and I'll be away from my printer for the next 5 days.  I will have remote access to the OctoPrint server if I can provide any additional logs or information.

foosel, 2017-12-04:
First of all, thanks a lot for the feedback so far!

On the checksum error, to just quote myself from #2262 to give all you an update (further handling in the dedicated ticket):

> So it looks like this got introduced by implementing #2226. I had to change up the communication error handling a bit in order to be able to even differentiate that particular error from others (since it needs different handling), and apparently while Repetier reports `wrong checksum` on a checksum mismatch, Marlin reports `checksum mismatch`, and that was no longer caught thanks to the change.
>
> Easy to fix in principle (I just need to add `checksum mismatch` to the list of recoverable communication errors). Now wondering though if there might be other variants out there... It definitely was easier to be able to just check for `checksum` but that was also wrong since that led to #2226 to begin with. Then again, the way the handling is implemented now, I might actually still be able to use that.
> 
> That it only arose in multi hour prints is a bit odd, but considering that I'm 100% sure what the reason for the error is here I feel fairly confident to think that's actually a coincidence.

@fiveangle 

The stock terminal filter regexes have been set to ``(Send: (N\d+\s+)?M105)|(Recv:\s+(ok\s+)?(B|T\d*):)`` for filtering temperature messages since 1.3.5 to accommodate that. But since those values are persisted in the config and *not* merged automatically, if you manually changed your terminal filters in any way (as in, you have an entry for them in your `config.yaml`) you won't have gotten those changed versions. I also put a note on that in the [1.3.1 release notes which was when the last change for the filtersbfore that occurred](https://github.com/foosel/OctoPrint/releases/tag/1.3.1), but completely forgot about it for 1.3.5 *cough*.

@goeland86

There's currently only [one plugin blacklisted at all, and only in two specific versions](). And those versions only caused problems with the UI to a degree that login and hence uninstall was no longer possible (this issue was actually what finally made me implement the blacklist feature).

Your issues sound like something really odd and alarming. Do you have any `octoprint.log` files to share of the unsuccessful start by any chance?

goeland86, 2017-12-04:
@foosel I will try to dig through the logs I have on the printer to find them tonight. But I may have accidentally run `rm -r .octoprint`. If I don't have them I will try to re-create the conditions that led to that.

However the UI not issues sound like what was going on there - it was basically timing out in the browser, even though OctoPrint had started and connected to Redeem (my on connect G-Code is a G28, and I heard the printer home after each and every restart of OctoPrint with systemctl).

foosel, 2017-12-04:
@goeland86 sorry, I was unclear - the issue with the UI wasn't as extreme as what you are describing. Stuff still loaded, server responded, but the JS in the plugin was buggy causing the client side JS of the UI to not initialize properly.

Also, I just realized I forgot to actually put the blacklist link URL into that comment - obviously Monday morning. http://plugins.octoprint.org/blacklist/ shows what's currently on the blacklist. Also edited the comment above so the link actually leads somewhere.

goeland86, 2017-12-04:
@foosel the UI timing out could've been slowness compounded by the beaglebone's wifi being slow (bad USB2 support with a wifi-G dongle), but I will try to get you those logs ASAP to confirm/deny the problem is the same.

ChrisHeerschap, 2017-12-05:
Have been working through the improvements on 1.3.6rc1 and everything's been working well for me - have completed many prints with no problems - but did find one particular issue with persistent options in the gcode tab. Have opened https://github.com/foosel/OctoPrint/issues/2267 for that issue.

bjarchi, 2017-12-05:
Running 1.3.6rc1 since release a few days ago on my sole printer, pretty heavy use since I updated, and I've noticed no issues so far.

goeland86, 2017-12-05:
@foosel I've been unable to reproduce the error I first got, and unfortunately the octoprint.log is gone. However I don't recall seeing anything other than a single error about incorrect file permissions on a .git file. 

I'll stand by my guess that it was super slow wifi compounded by the Beaglebone's limited resources when rebooting and restarting all services that caused the timeout on the browser side. The reason why is that I tried to reproduce that on a separate beaglebone connected over the USB-host bridge this time, which is usually more reliable than a USB network dongle  (different USB chips apparently, go figure!). It was still slow and as I was specifically restarting Octoprint only, there were more resources available for the startup process which avoided the timeout. I got the standard "Octoprint is starting" message until it was ready.

I will keep trying to reproduce the bug and get a log file out of it if I can.

foosel, 2017-12-05:
@malnvenshorn Oops, I almost oversaw the issue you reported about the temperature offsets. I opened a new ticket for that in #2268. Already fixed though.

@goeland86 Thanks for getting back on that! A pity about the log files, but if you currently can't reproduce, chances are good it indeed was more a temporary environment issue rather than a general problem.

foosel, 2017-12-05:
Marking as closed because 1.3.6rc2 has been [released](http://octoprint.org/blog/2017/12/05/new-release-candidate-1.3.6rc2/). Comments are still possible though.

New feedback ticket is #2269

closed 2017-12-05 by foosel