OctoPrint/OctoPrint #2260: [brainstorming] Preventing popup spam

ntoff, 2017-12-03:
When disabling plugins, or re-enabling them after testing or installing multiple plugins (or uninstalling them), you can end up with an entire screen full of popups telling you to restart. This is somewhat annoying when you have to disable all your installed plugins in order to test something, but you still need 3rd party plugins to work (so safemode is out).

It'd be nice to assign popups a tag, and if one tag is already active, no more popups of that nature will be allowed.

The idea comes from here: https://github.com/sciactive/pnotify/issues/228#issuecomment-217470039 and is something I put into my fan speed slider plugin, if you set a speed outside the range, you'll get a popup telling you "hey, I've adjusted the speed", but it then checks for that popup if you set another speed outside of the allowed range, so you'll only ever see one popup per speed type (above threshold and below are two separate events).

It'd be great if OctoPrint would go "oh look, the user already has one plugin restart popup, I won't put any more on the screen".

Or maybe have a dedicated notifications section for the "boring" stuff, something like windows 10's sidebar thing where you see app notifications, a sidebar popup in octoprint where all the boring popups go and you're able to just mass dismiss all of them. Meanwhile all the important ones like bugfixes, new versions, etc all get popped up in the current fashion (given they aren't nearly as frequent and are important).

malnvenshorn, 2017-12-03:
> Or maybe have a dedicated notifications section

I think that would be a great solution. Something like here on github with the bell on the top right, just with a dropdown to see all notifications and the possibility to discard them all with a single click.

This is how it looks like on the web frontend of my nas:

[image]

foosel, 2017-12-04:
I'd rather not hide notifications in general behind a notification area (and the navbar is somewhat full anyhow). From my experience it's tricky enough to get people to read important messages when they pop up right into their faces, it becomes downright impossible when you need them to go look for them first, even if that involves blinking red icons. 

So instead of hiding stuff, maybe reducing the amount of less important notifications should be the goal here. Some more intelligent grouping might indeed make sense.

I understand the "plugin was installed/reinstalled/uninstalled, please reload/restart" messages from the Plugin Manager are a pain point here. Anything else that jumps to mind?

ntoff, 2017-12-04:
I wasn't suggesting hiding the notification area, but rather making a sort of modal popup where notifications go, rather than having them clutter the entire screen. So when a message pops up, it makes a little drawer slide out from the side (or top, but side seems best) and all notices are constrained within that modal (but it is still automatically visible). Like the notification system windows 10 has. Then there could be a button atop said modal with "close all", so if a user is just removing a thousand plugins (or installing them), they can quickly dismiss all the notifications and the modal would then vanish.

>Anything else that jumps to mind?

Lots, but they're their own separate issue (not really notification spam, but still related to notifications, sort of, but not).

token47, 2018-01-07:
Maybe the popup could show a counter of messages to read and some other compact info. When new notifications arrive, it will only increase the counter. To read them, hover the mouse and a bigger window opens with the list in order. This will limit the pop ups to max 1 bubble, but still be there, and still has to be dealt with.

cmock, 2018-02-26:
One more data point: I'm currently setting up a new Raspberry with octoprint, so there's no printer connected. That led to "communication error" notifications being created about as fast as I could close them, preventing me from accessing the login dialog, thereby preventing me to get to the settings and fixing the problem...
[image]

foosel, 2018-02-26:
Why do you get communication errors of no printer is connected? There is nothing to communicate with that could error out here.

cmock, 2018-02-26:
Seems to be [octoprint-tft](https://github.com/mcuadros/OctoPrint-TFT) which tries to get octoprint to connect over and over.

But the point I was trying to make is that this shows how the window can get so filled with notifications that one doesn't get to important UI elements, such as the login dialog.

ntoff, 2018-02-27:
That's up to the plugin dev to handle though, I took care of that in my fan speed plugin by assigning a value to each popup so you'll only ever see one popup message of each kind no matter how many times you  spam that speed button with an over limit value. I'd bring that up with the TFT plugin developer and ask them to do something about it.

cmock, 2018-02-27:
It's not a plugin, it is a standalone application that uses the API. The notifications are generated by octoprint.

foosel, 2018-03-05:
FWIW, the plugin manager in particular will produce way less notification spam starting with 1.3.7:

[image]

LazeMSS, 2021-03-31:
I think this is handle by the new "listing" way of notifications and could be closed - @foosel ?