OctoPrint/OctoPrint #2261: [Request] Support for XON/XOFF flow control for Marlin hosts

fiveangle, 2017-12-03:
[Discussed before](https://github.com/foosel/OctoPrint/issues/450#issuecomment-342861370), Marlin 1.1 and 2.0 can support serial flow control.  As @foosel [pointed out](https://github.com/foosel/OctoPrint/issues/450#issuecomment-343081342), a glaring omission was there was no method for a host to test for this feature to allow easy-enable (and admittedly, direct testing of protocol support at connect time would be much more host-side dev effort and infinitely more prone to breaking unsuspecting targets).

With  https://github.com/MarlinFirmware/Marlin/pull/8634 and https://github.com/MarlinFirmware/Marlin/pull/8635, current Marlin will now report in the M115 response the status of _SERIAL_XON_XOFF_ support:

> Send: M115
> Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Printrbot Metal EXTRUDER_COUNT:1 UUID:9c85d1db-7066-4084-b9de-f16881c3cf68
> **Recv: Cap:SERIAL_XON_XOFF:0**
> Recv: Cap:EEPROM:1
> Recv: Cap:AUTOREPORT_TEMP:1
> Recv: Cap:PROGRESS:0
> Recv: Cap:PRINT_JOB:1
> Recv: Cap:AUTOLEVEL:0
> Recv: Cap:Z_PROBE:0
> Recv: Cap:LEVELING_DATA:1
> Recv: Cap:BUILD_PERCENT:0
> Recv: Cap:SOFTWARE_POWER:0
> Recv: Cap:TOGGLE_LIGHTS:0
> Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
> Recv: Cap:EMERGENCY_PARSER:0
> Recv: ok P15 B3

**Cap:SERIAL_XON_XOFF:1** if support for it is present.

If this flow control scheme were implemented on the host side in Octoprint, it could be used to reduce the sensitivity of realtime delivery of gcode commands to the target, and potentially reduce [draining of the planner on Marlin targets](https://github.com/foosel/OctoPrint/issues/450) that support it (current Marlin _SERIAL_XON_XOFF_ implementation requires 1024 minimum buffer size so could sustain many seconds of halted serial communications, even on the fastest printers).  Supporting this flow control should (in theory) dramatically improve SD card file transfer speed as well.

If implented, the suggestion made in #2148 might make good sense to do at the same time as serial-based AVRs appear plenty capable of supporting 1,000,000 with the current Octoprint implementation, but is certainly not a requirement.  

The _SERIAL_XON_XOFF_ feature is slowly being implemented for additional targets in Marlin ([the 32-bit Due, for example](https://github.com/MarlinFirmware/Marlin/pull/7936)).

luw, 2018-01-06:
Any idea for what speeds can be expected with this change?  I no longer use the raspberry pi/octoprint for streaming gcode due to quality issues, and moving a 10MB file over 250kbaud is too tedious, so it's pretty much a webcam.

I would be very interested in something that speeds up moving files to the mega 2560's SD card (running whatever marlin).

JohnOCFII, 2018-01-06:
> I would be very interested in something that significantly speeds up moving files to the mega 2560's SD card (running whatever marlin).

Best way to improve that speed is with a WiFi SD card like the Toshiba FlashAir: http://www.toshiba-memory.com/cms/en/products/wireless-sd-cards/FlashAir/flashair-wireless-sd-card-W04.html

NicholasTracy, 2018-10-21:
Almost 1 year later, where are we with this support? I've been patiently waiting to use flow control with my direct serial link between the RPi and Marlin. 

Thanks for the efforts!

foosel, 2018-10-22:
Take a look into the discussion in the attached PR #2346 - severe problems were found while trying to get this to work reliably with the current architecture and the current behaviour in firmware.

foosel, 2018-10-22:
Take a look into the discussion in the attached PR #2346 - severe problems were found while trying to get this to work reliably between the current architecture and the current behaviour implemented in firmware which made it not viable at the moment, at least not for general consumption.

NicholasTracy, 2018-10-25:
I did see that PR as well. So is this possible for specific and specialized situations like for those of us using the hardware UART on both the RPI and the mega based controller? I understand the reasons for not wanting to pursue this until a better solution is presented for general audiences because of compatibility with other firmware and hardware. I strictly use Marlin on a mega based board and an RPI3+ with hardware UART on both where flow control would shine. This would negate the issues with the USB controllers and USB to Serial converters. If there is any work around at all I would be happy to explore it, if nothing else, for the sake of proving the concept on limited hardware/firmware combinations.

foosel, 2018-11-05:
> So is this possible for specific and specialized situations like for those of us using the hardware UART on both the RPI and the mega based controller?

Possibly, but it needs some legwork still.

guestisp, 2019-05-18:
Just to be sure: `SERIAL_XON_XOFF` should be kept off, if I understood properly (and also because enabling it doesn't allow me to print anymore via octoprint), but what If I set the `RX_BUFFER_SIZE` to 1024 or more ? Will this improve reliability when printing via octoprint ? I can't increase `BLOCK_BUFFER_SIZE` due to low mem

fiveangle, 2019-05-23:
Serial flow control is not implemented in Octoprint (see Gina's link to further discussion on continued upstream serial driver issues making implementation challenging).

For your other questions, they are best asked on the Marlin tracker or 3D printing forums, but in short, ping-pong serial comms used between Marlin and OP today make the RX buffer increase not useful for any improvements. 

-=dave

tphan26363, 2020-04-24:
XON/XOFF supposed to be a hardware initiation.  It not really need a major rewrite in code.  Just make the comport initiation has an option to use XON/XOFF or not.  When time come.  It can be useful.  Seriously I don't understand why it  is so hard.  Every programming language has this build in