OctoPrint/OctoPrint #2263: Webcam streaming is OK, but snapshot has issues.

bookledge, 2017-12-04:
#### What were you doing?

I'm installing OctoPrint on raspberry pi.
all has setup. 
but timelapse remains.

#### What did you expect to happen?

snapshot URL test pass!

#### What happened instead?

after pressing the test button, I can see Popup windows like this
"
Snapshot test failed
Could not retrieve snapshot URL, please double check the URL
"
but, stream URL test is perfect.
and I can see the webcam stream at control tab. 

only snapshot test has failed. 

I searched that issue all over the world. even FAQ.
but I can't find the issues like me. 

#### Did the same happen when running OctoPrint in safe mode?

Same in safe mode

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5 (master branch)

#### Operating System running OctoPrint

Raspbian jessie

#### Printer model & used firmware incl. version

not connected yet

#### Browser and Version of Browser, Operating System running Browser

Mac safari

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2017-12-04:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

bookledge, 2017-12-04:
#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/1526369/octoprint.log)

#### Link to contents of terminal tab or serial.log

I didn't connect with my print yet.  
maybe that reasons, in my .octoprint/logs folder  there is no serial.log file

#### Link to contents of Javascript console in the browser

[Error] Failed to load resource: 파일이 존재하지 않기 때문에 ‘messages.json’ 파일을 열 수 없습니다. (messages.json, line 0)
[Log] Less has finished and no sheets were loaded. (less.min.js, line 13)
[Info] Starting dependency resolution... (packed_core.js, line 13379)
[Info] ... dependency resolution done (packed_core.js, line 13480)
[Info] Initial application setup done, connecting to server... (packed_core.js, line 13738)
[Info] Connected to the server (packed_core.js, line 11834)
[Info] Finalizing application startup (packed_core.js, line 13716)
[Info] Going to bind 24 view models... (packed_core.js, line 13625)
[Info] Did not bind view model (4) (packed_core.js, line 13666)
"CuraViewModel"
"to target"
"#wizard_plugin_cura"
"since it does not exist"
[Info] Did not bind view model (4) (packed_core.js, line 13666)
"SoftwareUpdateViewModel"
"to target"
"#wizard_plugin_softwareupdate"
"since it does not exist"
[Info] User AnetAm8 logged in (packed_core.js, line 2165)
[Info] ... binding done (packed_core.js, line 13700)
[Info] Application startup complete (packed_core.js, line 13712)
#### Screenshot(s)/video(s) showing the problem:

[image]

=> that's my config.yaml webcam setting

[image]

=> stream test OK. screen shot.

[image]

=> snapshot URL test fail screen shot

I have read the FAQ.

foosel, 2017-12-04:
Things to check:

  * Does this snapshot URL work and produce a picture when you paste it in your browser's address bar? If not - nothing that OctoPrint can do about that.
  * Is this snapshot URL reachable from the machine OctoPrint is running on? Because if not - that's the problem, the snapshot URL needs to be reachable by the server (= Pi?), not the client (= browser).

In general, this is a support request - it's an issue with your specific configuration, not a bug in OctoPrint (when the snapshot URL is configured correctly the test button and timelapse work just fine). Hence closing this - nothing to do from OctoPrint's side here. Comments are still open though.

In the future: For questions or other kind of help or support, please get in touch on the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2017-12-04 by foosel

bookledge, 2017-12-04:
thanks for your feedback.

when I put the snapshot URL in browser, it works.
as you mention, I connect my printer to raspberry pi. 
but, URL test fail continues. 

I think that it's not mjpg-streamer issues, but OctoPrint. 
I think it's something to do from Octo.. 
I don't know the way to recover it..

bookledge, 2017-12-04:
I solve the problem.
but yet it's mysterious. 

that's due to port forwarding of my router. 
I set up 8081 port for mjpg-streamer.
and I open that 8081 port in router, I close the 8080. port.

Occasionally I open 8080 port, All problem has solved.
but, when I access to 8080 port in browser.. there is nothing.
only 8081 show me the webcam view.

I thing... 'Octoprint test setting' has problem.
stream is Okay.
but snapshot URL test is using the 8080 port.
it's abnormal situation.

please tell me why?

zhoushenglin, 2019-12-16:
hi, I think you should close ip camera login authentication.