OctoPrint/OctoPrint #2264: Octoprint WebInterface    "connection Timeout" from browser

DanielStartec, 2017-12-04:
#### What were you doing?

Trying to connect to Octoprint WebInterface

#### What did you expect to happen?

I was expecting The webinterface display

#### Did the same happen when running OctoPrint in safe mode?

Not tried yet

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5 (master branch)

#### Operating System running OctoPrint

OctoPi  0.13.0

#### Printer model & used firmware incl. version

Leapfrog Creatr

FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 
FIRMWARE_URL:http://www.mendel-parts.com 
PROTOCOL_VERSION:1.0 
MACHINE_TYPE:Mendel 
EXTRUDER_COUNT:2 

#### Browser and Version of Browser, Operating System running Browser

Chrome  Versione 62.0.3202.94

#### Link to octoprint.log

https://pastebin.com/tNJp075L

DanielStartec, 2017-12-04:
Hi GitIssueBot , could you please provide me which infos did I miss?

foosel, 2017-12-04:
> #### Link to contents of terminal tab or serial.log
> 
> [On gist.github.com or pastebin.com. If applicable, always include if unsure or
> reporting communication issues. Never truncate.
> 
> serial.log is usually not written due to performance reasons and must be
> enabled explicitly. Provide at the very least the FULL contents of your
> terminal tab at the time of the bug occurrence, even if you do not have
> a serial.log (which the Contribution Guidelines tell you where to find).]
> 
> #### Link to contents of Javascript console in the browser
> 
> [On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
> always include if unsure or reporting UI issues.
> 
> The Contribution Guidelines tell you where to find that.]
> 
> #### Screenshot(s)/video(s) showing the problem:
> 
> [If applicable. Always include if unsure or reporting UI issues.]
> 
> I have read the FAQ.

ntoff, 2017-12-04:
When it's unreachable, try to ping the device or ssh into it, just to see if it's octoprint that's down, or whether your device running octoprint has fallen off the network. I frequently have my pi fall off the network so have a script running that checks the connection every minute and reconnects in the event of a failure.

DanielStartec, 2017-12-04:
Tried to ping:  "Destination Host Unreachable"
Tried to netscan all the ip of the network:  cannot find the octoprint pc (RPI 1 type B) IP

Additional Info: Sometimes during the printing the WebInterface refresh, but a second  later is again unreachable
When the prints end I could reach the webinterface again

DanielStartec, 2017-12-04:
@ntoff  are you telling me that is rpi issue?

foosel, 2017-12-05:
He's telling you its either an issue with your pi or rather with its network connection. If you can't even ping it anymore, it's not OctoPrint that's the problem here, it's your Pi not even being reachable on the network at all (and thus also none of its services, like SSH or OctoPrint for that matter).

That's what you need to solve, and the OctoPrint bug tracker is the wrong place for that.

I suggest getting in touch on the established support channels as linked by the bot up there. Or maybe you might have even more success on the raspberry pi forums in this case, since it's your whole Pi that's falling off the network. That's not an OctoPrint or even OctoPi specific problem, it's a network problem.

Closing this since it's not a bug, feature request or brainstorming ticket.

closed 2017-12-05 by foosel

DanielStartec, 2017-12-07:
thanks a lot @foosel and @ntoff for your answer.

how you told me that could be a raspberry connection issue, I tried and almost solved with a

 `sudo apt-get update`    and   `sudo apt-get upgrade`  of the rpi.

I hope that the next time I will think on the rpi as an issue and try to solve that first.

@ntoff thanks for your script suggestion, I will take it on my mind  but for now I hope the update and upgrade solved the issue.