OctoPrint/OctoPrint #2267: Issue with persistent options in gcode viewer in 1.3.6 RC1

ChrisHeerschap, 2017-12-05:
#### What were you doing?

Running a print and testing the new features in the 1.3.6 RC1 changelog. Testing the features that I could test, including the persistent options in the gcode viewer. Discovered that when I toggled on "Also show next layer" and reloaded, I saw the gcode size warning (with a null size value) underneath the gcode viewer 

To reproduce in my environment:

* During a print - go to the gcode viewer.
* Toggle on the first five options (adding the two that aren't on by default)
* Toggle on "Also show next layer"
* GCode viewer works as expected. Changing from tab to tab works as expected.
* Reload the page via CMD-R or clicking in the address bar and hitting enter, at which point I see the gcode viewer working as expected, but no options are checked under "Renderer options" and the size warning shows underneath that, saying "You've selected for printing which has a size of . Depending..."
* Reloading that page again works, but with "Also show next layer" turned off. (all other options are reverted to how they were saved)

The same doesn't happen when choosing "Also show previous layer" - that works as expected and persists across page reloads. This is only a problem with "Also show next layer".

TL;DR: Check "Also show next layer" and reload - gcode viewer acts screwy. Reload again and it goes back to working as expected but with "Also show next layer" unchecked.

#### What did you expect to happen?

Based on the changelog, reloading the window should have brought back the gcode viewer with the same options checked and functional. The "center viewport" and "zoom in" options persist across reloads, which I believe is the purpose of the new functionality.

#### What happened instead?

All options are shown unchecked and the size warning is given underneath the gcode viewer with null values.

#### Did the same happen when running OctoPrint in safe mode?

Yes - I restarted in safe mode and was able to recreate the issue with the same steps. Turning on "Also show next layer" and reloading would give the issues.

#### Branch & Commit or Version of OctoPrint

1.3.6 RC1

#### Operating System running OctoPrint

octopi 0.13.0 - but manually updated via apt update/upgrade since install.

#### Printer model & used firmware incl. version

Monoprice Maker Ultimate, stock firmware

#### Browser and Version of Browser, Operating System running Browser

Google Chrome Version 62.0.3202.94 (Official Build) (64-bit) on OS X Sierra 10.12.6

#### Link to octoprint.log

https://gist.github.com/ChrisHeerschap/4e3d9045267fab84245599b84d30091c

#### Link to contents of terminal tab or serial.log

Terminal output is filled with info about the current print, so might not be pertinent. serial.log is empty (although has the message about how to enable it as per 1.3.6 RC1 feature enhancement!)

Terminal output from a "Copy all" in the terminal window:
https://gist.github.com/ChrisHeerschap/0a44cb5d04821ecdad869245fa3fbad5

#### Link to contents of Javascript console in the browser

https://gist.github.com/ChrisHeerschap/4e19da6d229caf8766e68dca7b756aac

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2017-12-05:
Nice find! 👍 And I wish all issues where that easy to solve 😄 

Turns out that this was a bug that was always in there but never got triggered since the `showNextLayer` flag usually wasn't set during initialization. It being set though caused an attempt to read the length of an as of yet undefined array containing the model information, and that in turn caused the whole GCODE viewer view model to not be bound, leading to everything being unchecked and stuff being visible that didn't make sense. 

Should be fixed in the next RC thanks to above commit.

foosel, 2017-12-05:
1.3.6rc2 was released earlier today which should fix this issue.

closed 2017-12-05 by foosel

ChrisHeerschap, 2017-12-05:
Confirmed that this is fixed in RC2 - thank you!

commit 743d590d (2017-12-05): model might not be set yet

So don't try to read its length right away.

Closes #2267