OctoPrint/OctoPrint #2269: [RC Feedback] Feedback on the 1.3.6rc2 Release Candidate

foosel, 2017-12-05:
Please provide *general* feedback on your experience with the 1.3.6rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

### Note for people who already ran 1.3.6rc1, then downgraded back to 1.3.5

You'll need to do a tiny fix in your `config.yaml`, made necessary by the switch to `pip` as update method in 1.3.6rc1, our you won't be able to switch back again to Maintenance RCs to test this second RC. You'll need to remove the ``method: pip`` line under ``plugins.softwareupdate.checks.octoprint`` and then restart. 

You can easily do this `config.yaml` change via [the YamlPatcher plugin](http://plugins.octoprint.org/plugins/yamlpatcher/) using the patch string

    [["remove", "plugins.softwareupdate.checks.octoprint.method", ""]]

or via the command line using 

    sed -i -e "s/method: pip//g" ~/.octoprint/config.yaml

b-morgan, 2017-12-05:
I'm not seeing any printing  problems but I don't think the issue(s) with the Gcode Viewer are fixed yet. I'll capture some data after the current print finishes.

foosel, 2017-12-05:
@b-morgan which issue are you talking about? #2267?

b-morgan, 2017-12-05:
Not exactly. I started a print, the GCode Viewer was working (with Sync with job progress checked). I left to do other things and when I came back, the GCode Viewer screen was blank (print was still running).

It is a short print so I'm going to reboot everything, enable safe mode, print it again, and be more observant. Then I should have enough information to open another issue.

foosel, 2017-12-05:
That's an issue that I can't remember seeing a ticket for? 

Is this something that only started under 1.3.6* or was that something you also saw before? If the latter it is not a regression and will have to wait until 1.3.7 now.

b-morgan, 2017-12-05:
I haven't seen any GCode Viewer issues with 1.3.5 so I believe this is new behavior.

b-morgan, 2017-12-05:
The GCode Viewer in Safe Mode behaved itself. Guess I'll try again with plugins enabled.

ChrisHeerschap, 2017-12-05:
I've confirmed that https://github.com/foosel/OctoPrint/issues/2267 is fixed in RC2, thank you! I'll run through the other things to verify what I can.

b-morgan, 2017-12-05:
After reloading (to remove Safe Mode), the GCode Viewer is behaving itself... I'll report again if it (or anything else) is unusual.

chippypilot, 2017-12-05:
I've updated and done 3 small prints successfully. no issues to report

Lordxv, 2017-12-06:
I will test this ! What Marlin firmware should you at least be running ?

foosel, 2017-12-06:
@Lordxv no specific requirements with regards to Marlin firmware version. In fact, it should also still work just fine not-Marlin firmwares ;)

ChrisHeerschap, 2017-12-06:
Several small prints and one 70MB print and still no issues. Ran through other things that I could think of and it's working nicely.

Lordxv, 2017-12-06:
So i run OctoPrint: 1.3.6rc2 and still got some Error:Line Number is not Last Line Number+1 and 
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
The no "Recv" are inconsistent.  

It runs on a raspberry pi zero w. And an Anet A8 with Skynet3D (version ??? ). Should i update to OctoPrint: 1.3.5 (stable) ?

foosel, 2017-12-06:
Yes, please open a bug report. And please include all the logs that are requested in the ticket template because otherwise there's no way to look into this. Just from your description though it sounds rather like you might be running into some actual communication issues (line number errors and timeouts are not necessarily a bug in OctoPrint, there are many possible causes for them, and most of them aren't OctoPrint bugs but firmware, power, cabling, ...).

Lordxv, 2017-12-06:
Ok thank you

andrivet, 2017-12-07:
I have printed a few objects without problems (Wanhao i3 Plus with my own firmware based on Marlin 1.1.6). I do have troubles with timelapse but I think it is not related to this RC since I already had it before (I am not able to configure the snapshot URL correctly).

foosel, 2017-12-07:
I found a small issue (software update plugin persists too much, causing issues when downgrading and potentially more going forward). Also switched `octoprint.org` and `plugins.octoprint.org` over to https finally and therefore adjusted all the references to those. That should actually already have been done for the plugin repository, but with the blacklist now I decided it was really about time.

I'll probably push out a third RC with these two changes tomorrow. That should give us another weekend of testing time (which coincides brilliantly with me needing to print some present related things ^^), and if things then still look as rosy as they do now I guess it's time to think about a stable release come next week :)

@andrivet 

In what way does "not being able to configure the snapshot URL correctly" express itself? ;) Certainly doesn't sound like a problem with the RC if it's something you've been fighting with longer, but you made me curious.

> I do like a lot the new GCode vieweer

It's the same that's been in there since ~2013 🤔

chippypilot, 2017-12-08:
I think I'm seeing a change of behaviour on the Actual Bed temperature display. This is in both the chart and the table below. At the start of the print there is a delay of >20 seconds before this responds to the temperature from the printer. It is delayed in starting, but then tracks real time, not delayed.The other temps. target and actual are tracking OK. 

I'll raise a ticket later when I've done a restart and checked it's still there.

foosel, 2017-12-08:
That might be related to how your printer reports your bed temperature. I certainly can't confirm a general issue with bed temperature reporting here. But I know that several firmware variants out there will only output the temperature of the heater it currently waits for heatup on, which means until the blocking heatup of the hotend is done (`M109`) no other temperatures are reported by the firmware.

chippypilot, 2017-12-08:
Thanks for the explanation. I hadn't noticed that behaviour before.

foosel, 2017-12-08:
Marking as closed because 1.3.6rc3 has been released. Comments are still possible though.

New feedback ticket is #2276

closed 2017-12-08 by foosel