OctoPrint/OctoPrint #2270: Missing functionality

henrikkoch, 2017-12-05:
Updated from RC1 

Now the control tab page is garbaged.

[image]

And I can't login to the admin part

No graphs are shown on the front page

[image]

ntoff, 2017-12-05:
Please see here for the information that is required for a support ticket: https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report

Check the developer console (F12 on the keyboard I think), you might have a plugin that's taken out the UI. 

If you can SSH into the pi (assuming it's octoprint running on a raspberry pi) you can find all the logs that are required / asked for in the issue template, in `~/.octoprint/logs`. Please find them and post them, not much can be done to help you without the required information. Note there is a dot / period / full stop on the front of the octoprint folder name, it's a hidden folder.

kantlivelong, 2017-12-05:
You can launch OctoPrint in [Safe Mode](http://docs.octoprint.org/en/master/features/safemode.html) to determine if it is just a broken plugin.

henrikkoch, 2017-12-05:
Thanks for the commens about Safe Mode. One bad thing is that the Power Button doesn't have any functions when pressing it to select shutdown, reboot, restart or safemode.  Nothing happens when pressing it with the mouse

henrikkoch, 2017-12-05:
OK after several reboots - It now seems to work - I will check more

kantlivelong, 2017-12-05:
Please fill out the original ticket message as specified in https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report prior to any additional messages. Otherwise the ticket will be closed.

ntoff, 2017-12-06:
>One bad thing is that the Power Button doesn't have any functions when pressing it to select shutdown, reboot, restart or safemode. Nothing happens when pressing it with the mouse

If you've had to just pull the plug on your pi, I would start fresh and re-flash the image. You may not have done anything super bad to it, but at the same time just pulling the power on a linux OS can sometimes lead to file corruption.

In future, if you can't use the UI to shut down, log in via SSH and run `sudo shutdown -r now` to reboot or `sudo shutdown -h now` to shutdown and "halt" for power off.

foosel, 2017-12-06:
This screenshot just screams JavaScript error which would be trivial to confirm if you had provided the requested information. Now I fear this will remain a mystery 😕

foosel, 2017-12-07:
Closing due to lack of information.

closed 2017-12-07 by foosel