OctoPrint/OctoPrint #2273: [Request] Use M211 gcode to set bed size automatically.

eyal0, 2017-12-07:
I have read the FAQ.

Marlin firmware (and maybe others) is able to report the size of the printer from firmware:

```
Send: M211
Recv: echo:Soft endstops: On   Min: X0.00 Y0.00 Z0.00  Max: X230.00 Y290.00 Z340.00
Recv: ok
```

Perhaps have a button in the printer configuration that, when pressed, would send M211 to the printer, get the min and max, and automatically fill in the bed size and shape?

If you like the idea and need help with it or want me to do it, let me know.  If you think that you won't add such a feature, let me know and I'll drop it.

foosel, 2017-12-11:
Sounds like a neat idea. Would need to handle "not connected" state and missing return value though.
 I'm thinking something along the lines of "Attempt to read from printer", clicking sends M211, asynchronous event triggered on the comm layer on parsing such a line causes the data to be filled in in the frontend if the profile editor is open, maybe?

eyal0, 2017-12-12:
Yes, that's also how I think that it should work.  A button that, when pressed, tries to get the M211 information from the printer.  I think that when the button is pressed, it should override whatever settings the user already has filled in.

When it overrides the current settings, should those be saved or only modify the textboxes and save when the user presses save?

Say it fails because the firmware doesn't support it or the printer isn't connected or whatever.  In that case, the settings won't be copied over and the user will have no way to distinguish between failure to get settings and waiting for settings.  Potentially the user will sit there, waiting for it to finish, not realizing that it failed.  How do we prevent that?  One way is to make it synchronous, maybe?

What do you think?

foosel, 2017-12-13:
> When it overrides the current settings, should those be saved or only modify the textboxes and save when the user presses save?

I'd say the latter, maybe with some kind of flag indicating "read from printer"?

> How do we prevent that? One way is to make it synchronous, maybe?

Making anything synchronous will cause a lot of problems in the current architecture (you effectively block the server if you have on request sitting there and wait for an answer from the printer).

Just looking at the backend I'd go for a "send command" - "send event when answer is received" approach. And in the frontend pressing the button would establish a connection to the printer (if not already present - or do we limit to active connections?), send the command, mark itself as busy, then wait a (configured) timeout for the triggered event. If it comes, hooray, apply values. If it doesn't, too bad, maybe display that somewhere. In any case, mark as unbusy again (is that even a word? I'm sorry dear English). If it comes outside of active button - to be decided ;) (unsure whether to still apply or just silently ignore in that case, applying might have the downside that the user doesn't expect that anymore at this point).