OctoPrint/OctoPrint #2274: [Bug]: Username and password are sometimes visible in browser address barRe

andrewsil1, 2017-12-08:
#### What were you doing?
I have noticed that sometimes when relaunching a browser (Edge, Windows 10 in my case), that had a window from an earlier session of Octoprint, my username and password are visible in clear text in the browser address bar.  I know I've clicked "Remember Me" in the past so I do expect the info to be saved somewhere, but I definitely am not expecting cleartext representations to be easily visible to the user (and anyone else nearby.)

An example of the URL string would be:
http://octopi.domain.local/?username=xxxxxx&password=yyyyyy

#### What did you expect to happen?
Username and Password should be hashed or otherwise obfuscated, or passed via cookies, or something.

#### What happened instead?
The username/password were clearly visible in the UI in plaintext.

#### Did the same happen when running OctoPrint in safe mode?
I believe so.

#### Branch & Commit or Version of OctoPrint
Current release branch as of 12/7/2017
1.3.5 master.

#### Operating System running OctoPrint
Raspi Zero, Octopi 0.14.0

#### Printer model & used firmware incl. version

Felix Tec 4

#### Browser and Version of Browser, Operating System running Browser

Edge  41.16299.15.0

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log
#### Link to contents of Javascript console in the browser
#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ!.

foosel, 2017-12-08:
This is a duplicate of #2108 and fixed in the upcoming 1.3.6 release.

closed 2017-12-08 by foosel

andrewsil1, 2017-12-08:
Thank you - sorry for the dupe!