OctoPrint/OctoPrint #2275: Running Python 2.0 script in Python 3.0?

xmai77, 2017-12-08:
I was following the guide to setup Octoprint on Windows. I'm running Windows 10.

When running the command to start the server, I'm met with an error:

```
(venv) C:\OctoPrint>octoprint serve
Traceback (most recent call last):
  File "C:\OctoPrint\venv\Scripts\octoprint-script.py", line 11, in <module>
    load_entry_point('OctoPrint==1.3.5', 'console_scripts', 'octoprint')()
  File "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\__init__.py", line 405, in main
    from octoprint.cli import octo
  File "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\cli\__init__.py", line 135, in <module>
    from .dev import dev_commands
  File "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\cli\dev.py", line 234, in <module>
    @dev_commands.group(name="dev", cls=OctoPrintDevelCommands)
  File "C:\OctoPrint\venv\lib\site-packages\click-6.2-py3.6.egg\click\core.py", line 1169, in decorator
    cmd = group(*args, **kwargs)(f)
  File "C:\OctoPrint\venv\lib\site-packages\click-6.2-py3.6.egg\click\decorators.py", line 115, in decorator
    cmd = _make_command(f, name, attrs, cls)
  File "C:\OctoPrint\venv\lib\site-packages\click-6.2-py3.6.egg\click\decorators.py", line 89, in _make_command
    callback=f, params=params, **attrs)
  File "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\cli\dev.py", line 26, in __init__
    from octoprint.util.commandline import CommandlineCaller
  File "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\util\__init__.py", line 8
    from __future__ import absolute_import, division, print_function
    ^
SyntaxError: from __future__ imports must occur at the beginning of the file
```

I went into the "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\util\__init__.py"  script and moved the __future__ command on line 8 above the multi-line comment at the start of the script, but after the first __future__ line. This fixed the issue. However, when trying to run the script again I'm met with a different error: 

```
(venv) C:\OctoPrint>octoprint serve
Traceback (most recent call last):
  File "C:\OctoPrint\venv\Scripts\octoprint-script.py", line 11, in <module>
    load_entry_point('OctoPrint==1.3.5', 'console_scripts', 'octoprint')()
  File "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\__init__.py", line 405, in main
    from octoprint.cli import octo
  File "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\cli\__init__.py", line 136, in <module>
    from .client import client_commands
  File "C:\OctoPrint\venv\lib\site-packages\octoprint-1.3.5-py3.6.egg\octoprint\cli\client.py", line 166, in <module>
    @click.option("--str", "-s", "str_params", multiple=True, nargs=2, type=click.Tuple([unicode, unicode]))
NameError: name 'unicode' is not defined
```
I'm not much of a programmer but after some digging, it looks like it's an issue with 'unicode' being a Python 2 thing but I'm trying to run it in Python 3. The instructions to install Octoprint on Windows does have me install Python 3, so I'm at a bit of a loss.

ntoff, 2017-12-08:
>The instructions to install Octoprint on Windows does have me install Python 3, so I'm at a bit of a loss.

No, no it does not. The first step literally tells you to install the 32bit version of python 2.7

https://github.com/foosel/OctoPrint/wiki/Setup-on-Windows

>Install Python 2.7 for 32bit from here (choose the "Windows x86 MSI Installer" package). Make sure to include pip in the installation and also to have the installer add python to your path.

The git bot has kindly provided a link to the mailing list and to the google+ community, please don't use this bug tracker as a support forum.

foosel, 2017-12-08:
Not a bug, use Python 2.7 instead of 3 and stuff will work just fine.

closed 2017-12-08 by foosel