OctoPrint/OctoPrint #2278: Error upgradting to 1.3.6

RyuKajiya, 2017-12-12:
Update exits with an error:

Processing dependencies for OctoPrint==0-unknown
Searching for monotonic>=1.3,<1.4
Reading http://pypi.python.org/simple/monotonic/
Couldn't find index page for 'monotonic' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
No local packages or download links found for monotonic>=1.3,<1.4
error: Could not find suitable distribution for Requirement.parse('monotonic>=1.3,<1.4')

Since then Octopi is brocken, Web UI is only ASCII and can't log in any more.

foosel, 2017-12-12:
Try the steps outlined in this post: https://octoprint.org/blog/2017/11/01/octopi-0.12-pip-issues/. Your problem sounds like the exact same issue and is caused by an outdated pip version in your environment, not OctoPrint.

Moorviper, 2017-12-12:
Hi i also had this issue with OctoPi 0.13.
I followed the steps and it worked mostly fine.

```
sudo service octoprint restart
```
did not work for me.
But after a very long reboot I could access the web interface.

RyuKajiya, 2017-12-12:
Thanks this seems to have worked. Altho it is strange because up until 1.3.5 I never had any problems.

foosel, 2017-12-13:
The problem is a change that happened outside of OctoPrint in a part of the Python infrastructure it relies on (the pip command and how it interacts with the python package index). That change happened a couple days after 1.3.5 was released. I pushed out the above important announcement to make users aware of this - I suggest to always read those announcements and ideally also subscribe to the OctoBlog's RSS feed or at least follow OctoPrint3D on Twitter to not miss stuff like that.

foosel, 2017-12-13:
Keeping this open for a bit longer just in case more people run into this now. Marked as solved though.

cxt666, 2017-12-13:
Same issue as @Moorviper for me, and also the same solution.

foosel, 2017-12-13:
@Moorviper @cxt666 I'd like to know how you got this issue to arise on OctoPi 0.13. My final tests before pushing out a release involve flashing various cards freshly with both 0.13 and 0.14 and I certainly couldn't trigger that issue on 0.13, neither when it first arose updating to 1.3.5, nor later with 1.3.6rc1, rc2, rc3 and the final stable 1.3.6 yesterday. Any particular changes you did to the image after initial flash? Accidental downgrade of pip? 

In any case, I've [added another entry to the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#i-got-some-error-about-no-suitable-distribution-during-update-and-now-my-server-wont-start) that isn't 0.12 exclusive and contains the solution and updated the blog post as well.

RyuKajiya, 2017-12-13:
Is it no possible to include this in the update script?
Or do you have to upgrade OctoPi (haven't seen a FAQ on how to update OctoPi yet).

foosel, 2017-12-13:
Since this issue arose AFTER OctoPrint 1.3.5 was released, nope, no chance to include it in the update script shipped with that version. Otherwise I might have had a chance to do that, but not with this exact order of events. And adding such a workaround into 1.3.6 or 1.3.7 wouldn't make any sense now either, since in order to update to that you need to already have this fix applied.

It would also be a bit... wrong. Considering that it's not actually an issue with OctoPrint but rather the environment that OctoPrint just happens to run in. It would feel like overstepping the bounds and responsibilities of that application if it started reconfiguring your environment for you, unprompted.

I've thought briefly on keeping `pip` updated the same way as OctoPrint, but I'm still not convinced this might be the best idea for stability (there have been situations in the past where a newer version of `pip` actually broke something that was working previously).

RyuKajiya, 2017-12-13:
Not familiar with the techniques used here, can you add a check if a certain version of pip is installed and abort before updating anything with an error stating that it needs to be updated first?

Moorviper, 2017-12-14:
Maybe something was wrong with the SD card i don't shut the pi down when i turn off the printer.
i think there was a problem when i installed some Plugins before the last update ...
-----------------------------------------------------------------------------------------
Ich glaube bei mir war es eine Kombination aus Upgrade auf eine neue Stable und Plugin-Update/Installation (welches das war keine Ahnung) Ich hatte meine ich ein Plugin direct über Github per Console installiert, weil es noch nicht im offiziellen Repo war. Ich denke mal das könnte es gewesen sein.

closed 2018-12-10 by foosel