OctoPrint/OctoPrint #2280: Plugin update fail after new 1.3.6 Upgrade

alasharim, 2017-12-13:
#### What were you doing?

   1. I was updating to Octoprint 1.3.6
   2. I was also asked to update a Plugin called "FilamentManager" to 0.5.0

#### What did you expect to happen?
   1. I expected both the Octoprint and the plugin to be updated
   2. also expected for the plugin to work after reinstall but it did not and now I cant install it at all :(

#### What happened instead?
   - Octoprint got updated successfully but FilamentManager did not

#### Did the same happen when running OctoPrint in safe mode?
   - Plugins are disabled during safe mode

#### Version of OctoPrint

   - 1.3.6

#### Operating System running OctoPrint

   - OctoPi 0.14.0

#### Printer model & used firmware incl. version

   - Creality CR-10

#### Browser and Version of Browser, Operating System running Browser

   - Google Chrome Version 62.0.3202.94 (Official Build) (64-bit)
   - Windows 10
#### Link to octoprint.log

   - octoprint.log
     -  https://gist.github.com/alasharim/286808cd4c54cfd7cab34bf5f28fbe10
   - plugin_pluginmanager_console.log:
     -  https://gist.github.com/alasharim/edd5cfb8f355d75fe46925029afbab24

#### Link to contents of terminal tab or serial.log

   - Not Applicable.

#### Link to contents of Javascript console in the browser

   - contents_of_Javascript_console.txt
      - https://gist.github.com/alasharim/574cf4eeeee65f546597cd054b87aeea

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

Manichee, 2017-12-13:
https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/50

https://github.com/malnvenshorn/OctoPrint-FilamentManager#setup

foosel, 2017-12-13:
Closing because that's a problem with the plugin, not with OctoPrint. See the linked ticket.

closed 2017-12-13 by foosel