OctoPrint/OctoPrint #2281: [Request] start collecting data to estimate time from second layer

eubr, 2017-12-13:
In doing my first couple prints with octoprint, I'm noticing that the estimated time starts very high and then gradually gets more reasonable in the first part of the print.

In the print I'm doing right now, in the span of a couple of minutes, the estimate has gone from 4 hours to 3 hours.

I am assuming that the first, slowest layer, in combination with other initialization gcode, throws off the estimation. Would it be possible to add a setting that allows to start collecting the data for estimation from the nth layer to fix this?

foosel, 2017-12-13:
Tricky - because it's not trivial to determine what even IS the first, second, third, .. layer. Considering that there are things like z hops, spiral mode, priming routines that go through multiple z levels and such, it's not just "second z-height = second layer". Even with a file analysis before the print (which doesn't necessarily take place, depends on how impatient the user is after upload ;)) it's not trivial, without it becomes even more difficult since all you can look at is the file up to this point.

eubr, 2017-12-13:
Good point!
Here's an idea for a temporary fix for my particular situation I might be able to implement: since I always know the z coordinates of the second layer, would the plugin api (which I don't know yet), would I be able with a plugin to give some hints to the time left estimator? Is this exposed in some API? If it is, I'll try to make a plugin.

justinmrkva, 2017-12-14:
I've been tossing around the idea of calculating the print speed by taking the ratio of distance traveled at the current time compared to the total distance traveled for the whole print.

Basically, for each command, calculate (length/speed) the movement and sum the values. While printing, compute the current value of this metric, and combining it with the current time should give a fairly accurate approximation. Using length/speed for each segment takes into account the speed of each movement, which also means it would correct for the speed decrease near the top of narrow prints as well as the slower first layer. For commands where the speed is unknown, you could just use some heuristic (like the average speed of the rest of the moves) instead.

Bear in mind this is just a casual thought and there are a lot of subtleties that would need to be addressed, but it would most likely be a pretty good estimate in a lot of cases.

foosel, 2018-05-17:
OctoPrint 1.3.9 will ship with two new hooks [`octoprint.filemanager.analysis.factory`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-filemanager-analysis-factory) and [`octoprint.printer.estimation.factory`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-printer-estimation-factory) that will allow plugins to override both the initial gcode analysis implementation as well as the live print time estimation. 

That should allow people to experiment with this stuff and find better but possibly more hardware/workflow/setup specific approaches that provide more accurate results than a stock option that has to *always* work can.

Knock yourselves out with this ;)