OctoPrint/OctoPrint #2282: Problem with blacklist plugin wizard on Android tablet

ir-fuel, 2017-12-13:
#### What were you doing?

Installing 1.3.6 on Raspberry Pi

Ideally provide exact steps to follow in order to reproduce your problem:

Click on "Update" in OctoPrint

#### What did you expect to happen?

1.3.6 should work

#### What happened instead?

Stuck on the blacklist plugin config screen

#### Did the same happen when running OctoPrint in safe mode?

No idea

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

Lenovo Tablet

#### Link to octoprint.log

N/A
#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

No idea how to do this on tablet

#### Screenshot(s)/video(s) showing the problem:

[image]

The problem only happened on my crappy Android tablet (Lenovo Tab 10. I bought it only for OctoPrint). On my computer all looks fine: 

[image]

ntoff, 2017-12-14:
Try turning it off and back on again (no really). I find my stupid android tablet's browser likes to cache old pages forever, maybe rebooting the tablet will help.

foosel, 2017-12-18:
As I'm wrapping things up for 2017, I'll take a look at this come January! 🎄

Can't promise at all though that I'll be able to solve this, let alone figure out what the problem is to begin with. Sounds more like an issue with the mystery browser on that tablet than anything else.

foosel, 2018-06-18:
Still leaning towards mystery browser issue. I can't really work around any browser problem out there, so I'm closing this as "won't fix" since I can't look into it. If someone capable of reproducing this wants to send a PR though that solves the issue without causing compatibility problems with other browsers, I'll happily merge that.

closed 2018-06-18 by foosel