OctoPrint/OctoPrint #2283: Print temperature always sets to 175 degrees

14422, 2017-12-13:
#### What were you doing?

1. Slice any model with printing temperature of 194 degrees for example
2. Press "Print with Octoprint"
3. Printing begins normally but target temperature is set to 175 degrees

It can be manually corrected but at each new print, it's set to 175 degrees again.

#### What did you expect to happen?
Target temperature to be 194 degrees

#### What happened instead?
Target temperature is always 175 at each new print

#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint
OctoPi 0.14.0 on Raspberry Pi 3

#### Printer model & used firmware incl. version
Prusa i3 MK2 with latest 3.1.0 firmware

#### Browser and Version of Browser, Operating System running Browser
Chrome 63.0.3239.84 on Windows 10 Professional 64B

#### Link to octoprint.log
N/A. Can't easily access my Raspberry now and I'm tired of reporting this already. No offense but this has been the most tedious bug report I have ever needed to do.

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

ntoff, 2017-12-14:
>N/A. Can't easily access my Raspberry now and I'm tired of reporting this already. No offense but this has been the most tedious bug report I have ever needed to do. Everything worked in the previous version of Octoprint so can't see the problem being at least in my hardware.

No offence but they're tedious for a reason.

What do you want us to do? Play 20 questions to get the info out of you needed to fix any issues? The logs are requested for a reason, to see whether OctoPrint is sending the correct data to the printer, and the printer is just ignoring it. Or whether OctoPrint is sending the wrong data, or whether a user setting or plugin is interfering somewhere.

Please provide ***all*** the requested information.

14422, 2018-01-11:
Updated Pastebin logs of the problem. Terminal log is up to a stage where printing has started with nozzle temperature set to 175 temperature (193 in Cura).

Javascript console log from Chrome is also taken at the same spot.

ntoff, 2018-01-11:
There are no references to a temperature of 193 in your terminal log, only 175, I'd say octoprint is functioning just fine, looks like cura is broken and not outputting the correct gcode.

Probably similar to this issue: https://www.reddit.com/r/3Dprinting/comments/7nod77/octoprint_temperature_issues/ I suggest trying a different version of cura just to be sure.

14422, 2018-01-11:
Downgraded to Cura 3.0.4. Everything works again.

foosel, 2018-01-11:
So can this be closed then?

14422, 2018-01-11:
Yes.

foosel, 2018-01-11:
Perfect, thanks!

closed 2018-01-11 by foosel