OctoPrint/OctoPrint #2284: GCODE viewer no longer keeps what its printed on the viewer

josemenendezferreira, 2017-12-13:
I've discovered to my surprise that the GCODE viewer no longer works.

Its not a huge issue, because... who sits there watching the gcode lines being drawn on the browser (I do) but now all I get are small traces that show up and disapear instantly.

Tried checking all the options, center, zoom, show previous, show next, different browsers (chrome and firefox) nothing...

Like I said, its not a huge issue but I think it should be looked at.

Thanks in advance for your time and effort.
Sincerely,
-Orallo

I have read the FAQ.

foosel, 2017-12-14:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

josemenendezferreira, 2017-12-14:
Hi!

Sorry, I'm new at reporting bugs and I have to learn all the etiquette!  My apologies!

Version: 1.3.6 (latest and greatest! I hadnt seen this behaviour before!)
Logs: Attached
[octoprint.log.zip](https://github.com/foosel/OctoPrint/files/1561014/octoprint.log.zip)
JS Console:
This page includes a password or credit card input in a non-secure context. A warning has been added to the URL bar. For more information, see https://goo.gl/zmWq3m.
packed_core.js?5d8232e6:14276 Starting dependency resolution...
packed_core.js?5d8232e6:14386 ... dependency resolution done
packed_core.js?5d8232e6:14689 Initial application setup done, connecting to server...
packed_core.js?5d8232e6:12672 Connected to the server
packed_core.js?5d8232e6:14667 Finalizing application startup
packed_core.js?5d8232e6:14551 Going to bind 32 view models...
packed_core.js?5d8232e6:14604 Did not bind view model CoreWizardAclViewModel to target #wizard_plugin_corewizard_acl since it does not exist
packed_core.js?5d8232e6:14604 Did not bind view model CoreWizardPrinterProfileViewModel to target #wizard_plugin_corewizard_printerprofile since it does not exist
packed_core.js?5d8232e6:14604 Did not bind view model CoreWizardWebcamViewModel to target #wizard_plugin_corewizard_webcam since it does not exist
packed_core.js?5d8232e6:14604 Did not bind view model CoreWizardServerCommandsViewModel to target #wizard_plugin_corewizard_servercommands since it does not exist
packed_core.js?5d8232e6:14604 Did not bind view model CoreWizardOnlineCheckViewModel to target #wizard_plugin_corewizard_onlinecheck since it does not exist
packed_core.js?5d8232e6:14604 Did not bind view model CoreWizardPluginBlacklistViewModel to target #wizard_plugin_corewizard_pluginblacklist since it does not exist
packed_core.js?5d8232e6:14604 Did not bind view model CuraViewModel to target #wizard_plugin_cura since it does not exist
packed_core.js?5d8232e6:14604 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?5d8232e6:2267 User Orallo logged in
packed_core.js?5d8232e6:14643 ... binding done
packed_core.js?5d8232e6:14663 Application startup complete

Hope this helps!

I've tried with Chrome and Firefox, both on normal and incognito modes... Same behaviour.  The Gcode viewer sketches traces of print and they instantly disapear.

I'm attaching a screen cap too so you can see what I see.
[image]

Thanks for all your time and effort!

-Orallo

foosel, 2017-12-15:
Please restart OctoPrint in safe mode and see if the same thing happens there as well. Also please provide a gcode file with which you are seeing this.

foosel, 2018-01-11:
@josemenendezferreira Did you try the above? Also still waiting for the file.

closed 2018-01-22 by foosel