OctoPrint/OctoPrint #2288: Octoprint and ultimaker 3

JPFrancoia, 2017-12-14:
I (my research group) bought a new Ultimaker 3 printer. As you might know, the UM3 is quite different from UM2, and it doesn't have USB type B. Which means I can't connect it to print from Octoprint (at least not easily).

My question is quite simple:
Is there a way to connect a UM3 to octoprint ? Has someone done it before ?

Please don't tell me I should have bought a UM2. We already have five UM2+, all connected to a Pi running octoprint.

I'm willing to help (code, bug report, whatever) to make it work.

Also I wonder if it's possible to install Octoprint **directly** on the UM3. It runs linux and has a ssh server once the developer mode enabled. Here is the partitioning of a UM3:

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       478M  389M   76M  84% /
devtmpfs        504M     0  504M   0% /dev
tmpfs           505M     0  505M   0% /dev/shm
tmpfs           505M  1.4M  503M   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           505M     0  505M   0% /sys/fs/cgroup
tmpfs           505M     0  505M   0% /var/cache/apt
tmpfs           505M     0  505M   0% /var/lib/apt
tmpfs           505M  4.0K  505M   1% /tmp
/dev/mmcblk0p3  3.2G  249M  2.9G   8% /home/ultimaker/tmp
```

UnacceptableUse, 2017-12-14:
Does the UM3 have any serial connections? If it does, perhaps the built in linux server is connecting over serial to the actual printer controller. Which would make connection probably easy. If not, it would certainly be interesting to try and get octoprint running locally via some sort of plugin.

JPFrancoia, 2017-12-14:
> Does the UM3 have any serial connections? If it does, perhaps the built in linux server is connecting over serial to the actual printer controller. Which would make connection probably easy.

I don't think so. Here is the connectivity of UM3:

[connection.pdf](https://github.com/foosel/OctoPrint/files/1559503/connection.pdf)

> If not, it would certainly be interesting to try and get octoprint running locally via some sort of plugin.

I volunteer to test/debug (mainly after january). Just point me in the right direction.

werweiss, 2018-07-12:
Hi guys,

isn't it possible to connect the Ultimaker 3 (UM3) by network to the Octoprint (New programming/driver) required. The Pi has Wifi and the UM3 also (or cable).
I want to have a big touchdisplay on my UM3 like the UM S5 has. Is that a possibility?

Thanks for your comments and thoughts,
Ich

gargahcerk, 2018-08-15:
Does anybody found a way to use Octoprint with Ultimaker 3?

huseyinkocakusakli, 2019-10-18:
Hey, does anybody found a way?

fieldOfView, 2019-10-18:
Do not get your hopes up, even if there is an open issue for it. The UM3, S5 and S3 are not designed to be used with an external print-host. They have a print-host built in that basically replaces OctoPrint, but is not open source so there will not likely be any plugins that add to or change its functionality.

The API exposed by the print host of the UM3/S5/S3 does not support sending gcode commands to the printer on at a time, nor does it expose the feedback from the printer on each processed gcode command while printing. OctoPrint needs that.

In theory, you modify the electronics to connect a board like the raspberry pi directly to the "motion controller" that is in the UM3/S5/S3 (bypassing the Olimex linux board that is in the printer). But then it would no longer work with Cura Connect, and you would have to reimplement a lot to get dual extrusion, material detection, bedleveling etc working again. In reality, it is just not possible.

huseyinkocakusakli, 2019-10-18:
Many thanks for information.

closed 2020-08-28 by JPFrancoia