OctoPrint/OctoPrint #2289: Problem updating from 1.3.5 to 1.3.6

Zblay, 2017-12-14:
#### What were you doing?

I was printing normally with octoprint 1.3.5 and then i wanted to upgrade to 1.3.6 

I don't know if there's anything anyone can do here other than telling me "reinstall a fresh version" but i'd like to avoid that and understand what went wrong. But i really am NO programmer. 

I tried to give as many details as i could. 

#### What did you expect to happen?
"You have successfully updated octoprint"

#### What happened instead?
And the server told me the update went wrong.
So i rebooted system, and now i have : 
 "503 Service Unavailable No server is available to handle this request. "

#### Did the same happen when running OctoPrint in safe mode?

I didn't try cause i cannot launch safe mode, or i don't understand how to do that through SSH

#### Version of OctoPrint

1.3.5

#### Operating System running OctoPrint
OctoPi v0.12.0

#### Printer model & used firmware incl. version
Prusa i3 hephestos running Marlin 1.1.6
https://github.com/Zblay/Marlin-1.1.6-Adapted-For-Prusa-I3-Hephestos- 

#### Browser and Version of Browser, Operating System running Browser
Mozilla Firefox V57.0.2

#### Link to octoprint.log
https://github.com/Zblay/Octoprint-problem-updating-1.3.5-1.3.6/blob/master/octoprint.log  

#### Link to contents of terminal tab or serial.log
https://github.com/Zblay/Octoprint-problem-updating-1.3.5-1.3.6/blob/master/serial.log 

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

I couldn't take any

I don't know if anyone can help, but well, it's worth a try, have a marvellous day !

Zblay, 2017-12-14:
-_- 

I don't understand what information is missing, could a human help me out ?

foosel, 2017-12-14:
You deleted the last line from the template and that is why the bot is complaining.

Anyhow:

> OctoPi v0.12.0

You might want to read and follow the steps in  https://octoprint.org/blog/2017/11/01/octopi-0.12-pip-issues/

It's just a shot in the dark here, but you probably got an error during update that some dependencies weren't found. You'll need to login via SSH and follow what's written in this blog post, your version of OctoPi is so old that some of the included python tools are no longer compatible to the current python infrastructure

Zblay, 2017-12-14:
Yes ! i remember it saying it didn't found a file ! 
I'll give it a try, right now, thanks A LOT.

Zblay, 2017-12-14:
Ok, there is progress, no more "503 service unavailable", now i get : 
"Looks like something went wrong during startup, the server is gone again. You should check octoprint.log."

Here is my updated log file : 
https://github.com/Zblay/Octoprint-problem-updating-1.3.5-1.3.6/blob/master/octoprint.log 

Thank you SO MUCH for helping

Zblay, 2017-12-14:
OK, after a second reboot, everything worked fine : i had a wizard asking me if i wanted to enable the plugin BlackList, i said yes, it seems to work.

Thank you so much, very sorry that i wasn't able to figure this out by myself.

Zblay, 2017-12-14:
It is printing right now. Issue can be closed. 
Thanks again

closed 2017-12-15 by foosel