OctoPrint/OctoPrint #2291: Printed failed - Communication error - checksum mismatch / timeout

JohnOCFII, 2017-12-15:
----

#### What were you doing?

I was about 90 minutes into a 200 minute print.  

I've been printing many similar pieces (the tree pieces the the Advent Tree: https://www.thingiverse.com/thing:2670389) over the last week without issue.  Not sure if this is the same issue that was resolved in the RC, or something different.  I had this issue at that time, but have printed for hours successfully since that time with the later RC, and now the release version of 1.3.6.

#### What did you expect to happen?

I expected print to finish properly.

#### What happened instead?

Print failed with hot-end staying in place above model, bed and hot-end heaters still on.

#### Did the same happen when running OctoPrint in safe mode?

Not attempted.

#### Version of OctoPrint

OctoPrint 1.3.6

#### Operating System running OctoPrint

OctoPi 0.13.0

#### Printer model & used firmware incl. version

Prusa i3 MK2 with firmware 3.0.12 w/Linear Advance

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

https://gist.github.com/JohnOCFII/c042ec4f801b2c36a0f83fbe5e152e14

#### Link to contents of terminal tab or serial.log

No serial.log, but here is the terminal window contents:

https://gist.github.com/JohnOCFII/780df83a53db40d484265880a06599a7

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2017-12-15:
I think this is a duplicate of #2285 and caused by the same firmware bug (no ok after a resend). Try enabling the workaround as described in the other ticket. 

And I have to open a bug in the prusa firmware repo about this I fear...

JohnOCFII, 2017-12-16:
Thanks, Gina.  I've enabled the workaround, and will try again.  The firmware I'm using, as well as the latest for the MK2 and the MK3 all have the new-ish linear advance code, which is something that firmware 3.0.12 (the release version) and earlier did not have.  I wonder if this behavior slipped in at that time.

foosel, 2017-12-18:
FWIW, I've opened a ticket at prusa3d/Prusa-Firmware#331 about this.

foosel, 2017-12-18:
I'm closing this since I've just reopened #2285 and this one is a duplicate of that

closed 2017-12-18 by foosel