OctoPrint/OctoPrint #2293: Octoprint not working Original Prusa i3 MK3 via GPIO

KissmanCZ, 2017-12-15:
#### What did you expect to happen?

Working Octoprint on Rpi Zero via GPIO with EINSY Rambo in Prusa MK3

#### What happened instead?

Nothing

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Original Prusa MK3, fw. 3.1.1 RC1 b122

#### Link to octoprint.log

https://pastebin.com/uKhXcJLJ

#### Link to contents of terminal tab or serial.log

https://pastebin.com/MvKm0gQL

foosel, 2017-12-16:
It works flawlessly over USB (I just tried), so that's not a general compatibility issue. It doesn't look like there's a printer on the other end of your `/dev/ttyAMA0` there, which unless you took further steps than just hooking up your Pi to the board wouldn't surprise me: by default, that's the Raspberry Pi's own console (RPi1, 2, 0) or the bluetooth serial (3, 0W, here GPIO serial is `/dev/ttyS0`), NOT anything you've connected to its GPIO ports. That's nothing OctoPi or OctoPrint specific but rather a general Raspbian behaviour.  

[Make sure you've enabled GPIO serial](https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/) and are trying to connect to the correct port. Then please report back.

KissmanCZ, 2017-12-16:
I disabled bluetooth from ttyAMA0 and console login from UART first of course :) tested with diferent speeds:

`Connecting to: /dev/ttyAMA0Changing monitoring state from 'Offline' to 'Opening serial port'Connected to: Serial<id=0xac95c730, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorStarting baud rate detectionChanging monitoring state from 'Opening serial port' to 'Detecting baudrate'Trying baudrate: 250000Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 125200Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 115200Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 230400Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 57600Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 38400Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 19200Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Trying baudrate: 9600Send: N0 M110 N0*125Baudrate test retry: 4Send: N0 M110 N0*125Baudrate test retry: 3Send: N0 M110 N0*125Baudrate test retry: 2Send: N0 M110 N0*125Baudrate test retry: 1Send: N0 M110 N0*125Baudrate test retry: 0Send: N0 M110 N0*125Changing monitoring state from 'Detecting baudrate' to 'Offline'Changing monitoring state from 'Offline' to 'Error: No more baudrates to test, and no suitable baudrate found.'Connection closed, closing down monitor`

KissmanCZ, 2017-12-16:
Tested to open serial on Rpi with sudo screen /dev/ttyAMA0, so I know, my port is right and not busy with another service or app. 

OctoPrint reports this when I open port on Rpi screen - port busy :)

`Connecting to: /dev/ttyAMA0Changing monitoring state from 'Offline' to 'Error: Connection error, see Terminal tab'Unexpected error while connecting to serial port: /dev/ttyAMA0 OSError: '[Errno 16] Device or resource busy: '/dev/ttyAMA0'' @ comm.py:_openSerial:1931 (hook default)`

And when I close screen /dev/ttyAMA0, i get

`Connecting to: /dev/ttyAMA0Changing monitoring state from 'Offline' to 'Opening serial port'Connected to: Serial<id=0xac95c730, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False)`

and again, no M110 respond :(

buzztiaan, 2017-12-16:
Does you octoprint user even have access rights to /dev/ttyAMA0 ?

KissmanCZ, 2017-12-16:
Yup... added "Pi" user under which service OctoPrint running to group "dialout"

pi@octopi:~ $ ls -l /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 64 Dec 16 19:23 /dev/ttyAMA0

buzztiaan, 2017-12-16:
and rebooted the pi? (or fully restarted all relevant processes?)

KissmanCZ, 2017-12-16:
Of course :) rebooted

KissmanCZ, 2017-12-16:
Maybe this is related? https://github.com/prusa3d/Prusa-Firmware/pull/297

foosel, 2017-12-16:
Well, I can certainly tell you that I also don't get any reply or even output at all using minicom, so whatever it is, it's not the fault of OctoPrint. My money would be on the firmware not being ready yet.

Comparison... minicom via USB, printer resets and firmware says hello:

```
Welcome to minicom 2.7

OPTIONS: I18n
Compiled on May  7 2017, 05:18:49.
Port /dev/ttyACM0, 21:01:11

Press CTRL-A Z for help on special keys

start
     echo:echo: Last Updated: Nov 29 2017 02:01:00 | Author: (none, default config)
                                                                                   Compiled: Nov 29 2017
                                                                                                        echo: Free Memory: 1522  PlannerBufferBytes: 1392
                                                                                                                                                         echo:Hardcoded Default Settings Loaded
                                                                                                                                                                                               crashdet_enable
CrashDetect ENABLED!
PAT9125_init:1
              FSensor
                      ENABLED

                             fsensor_enable

                                           PAT9125_init:1
                                                         tmc2130_init mode=0
initial zsteps on power up: 1016
initial zsteps after reset: 1016
echo:SD card ok
```

minicom via `/dev/ttyAMA0` after disabling bluetooth and getty, nothing at all happens:

```
Welcome to minicom 2.7

OPTIONS: I18n
Compiled on May  7 2017, 05:18:49.
Port /dev/ttyAMA0, 20:59:08

Press CTRL-A Z for help on special keys
```

buzztiaan, 2017-12-16:
so looks like nothing is connected to AMA0 ... tried swapping tx and rx
yet? :D

foosel, 2017-12-16:
Heh, good idea, but nope, that's not it either.

KissmanCZ, 2017-12-16:
So, we must wait for Prusa EINSY firmware fix as I understand?

foosel, 2017-12-16:
Based on my own observations so far: yes. If it wasn't late and I wasn't also fighting off the start of another cold I'd give it a try to simply compile a new firmware including the PR you linked and see where that gets us. But maybe someone else wants to try that ;)

In any case - closing because this doesn't look too be an OctoPrint problem at all but rather a MK3 problem. Happy to reopen should new information come in that proofs the contrary. Comments are still open though.

closed 2017-12-16 by foosel

KissmanCZ, 2017-12-16:
Ok... fast chat with Prusa support :(

`Já22:49
Hello, when will be firmware for MK3 with working UART feature availaible (Rpi Zero mounted to EINSY)?

Shane22:51
they are currently working on writing the firmware to support the pi0 board,

Shane22:51
I dont have an ETA yet`

KissmanCZ, 2017-12-16:
Thanks for your time Gina! ;)

foosel, 2017-12-16:
Ah, that at least confirms it then 👍

KissmanCZ, 2017-12-16:
There is a mention on Prusa forum before few minutes...

> Yes, I'd come to the same conclusion, so I tried building the current code from GitHub. It worked, and the internal Pi Zero W is now working as it should. However, I 100% agree with you that this is not necessarily the smartest/safest thing to do as lots of things are changing right now. I'll likely revert to the current release.

KissmanCZ, 2017-12-17:
New firmware is up ;) 3_1_1_rc2_b137 going to test.

Pretender1988smg, 2018-01-09:
Did you get it work? and when how?

markim, 2018-03-05:
Just want to put this here in case anyone else runs into problems: there are additional settings that must be made to the octoprint software before it will work with the MK3 and the Raspberry Pi Zero 0 https://help.prusa3d.com/l/en/article/Loz15FAgEk-octo-print-raspberry-pi-zero-w

foosel, 2018-03-05:
As a side note, contrary to Prusa, I do *not* recommend the RPi0W. It's WIFI interface has some serious issues and utilizing it causes severe CPU load. And it's overall rather slow.

markim, 2018-03-05:
Just to clarify: Is this solely because of the video streaming? I'm not having a bad time using mine to print so far. See top here: https://i.imgur.com/jZtyc1o.png I'm going to try using the camera module in a couple days. I don't plan on utilizing streaming, though, I just want timelapses. I feel like the RPiZW should have enough power to at least run serve the http and take a picture every every few minutes or so. Is the streaming only "turned on" when a user tries to connect to it, or can disable it? By the way, thank you for all your hard development work!

foosel, 2018-03-05:
Video streaming, or any kind of network utilization really. I got reproductions without running OctoPrint or a webcam at all by just copying a file from or to a location on my local network via the Pi0W. I'd also strongly advise against uploading (or downloading, e.g. by switching to the GCODE viewer) large files while printing. Anything that produces traffic on the network interface causes load and can cause stuttering. Same load issue on the Pi3, but with its four course it simply doesn't have a noticeable impact on print performance.

For some shots of `top` that show this, see [my comments here](https://github.com/guysoft/OctoPi/issues/318#issuecomment-284762963).

Just leaving the stream URL unconfigured and only limiting yourself to snapshots could work. Or using the SnapStream plugin (a good interval would need to be tested). But sorry, I really can't recommend it.

joysfera, 2018-03-11:
I'm tempted to attach an Ethernet NIC to RPi0 (via SPI) and give it a try. If the CPU utilization wasn't that scary it could be a good compromise for Octoprint.

n1cho1as, 2020-01-31:
> As a side note, contrary to Prusa, I do _not_ recommend the RPi0W. It's WIFI interface has some serious issues and utilizing it causes severe CPU load. And it's overall rather slow.

I bought a PI Zero W without making sure it would work 100%. My bad for trusting Prusa. I want to use this card so that I can perform bed leveling using one of the add-ins. If I disable the WiFi component, will this card be able to perform this basic function? Would I just add a USB WiFi or ethernet dongle to this device?

arhi, 2020-01-31:
> card be able to perform this basic function

Should be. It's not a board you wanna use to print from but octoprint without any serious plugins (the bed level visualiser etc will not be a problem) it will work... so you setup the rpi0w with octoprint, just make sure you do not print from the octoprint but you print from the printer's sd card, so even if rpi0w cpu crap out it's not affecting the printer... this will make bunch of plugins unusable (you cannot do octolapse for e.g.) but it will still allow you to monitor the printer and use few interesting plugins... just don't print from it as it will ruin your prints

n1cho1as, 2020-01-31:
> Should be. It's not a board you wanna use to print from but octoprint without any serious plugins (the bed level visualiser etc will not be a problem) it will work... so you setup the rpi0w with [...]

Thanks for the advice. I didnt plan on using it for printing, but good to know about the issues as I'll steer clear using it as a print server. I'm awaiting on the long edge connectors and looking at trying to connect the rpi0w via the USB interface. I've not played with a PI before, so hopefully it wont be too hard connecting via USB rather than the ENSY edge connectors.

n1cho1as, 2020-01-31:
> > card be able to perform this basic function
> 
> Should be. It's not a board you wanna use to print from but octoprint without any serious plugins (the bed level visualiser etc will not be a problem) it will work... 

Thanks for the advice. I have since installed octoprint (not the PrusaPrint version) using the steps in the prusa gude. Really configuring WiFi and removing "console=serial0,115200" from /boot/cmdline.txt. Having zero knowledge with the PI and OctoPrint I have no idea that edit was needed.  I have connected the rpi0w to my i3 MK3s via USB and can connect to the printer using the OctoPrint web server. I also installed the auto bed leveling plug in. The issue I see now is that OctoPrint keeps loosing it's connection to the printer and it's almost like the printer is resetting. Is this just a configuration issue or related to the rpi0w?

This is my terminal output:
```
Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Changing monitoring state from "Detecting serial port" to "Opening serial port"
Connected to: Serial<id=0xb21f44f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 115200
Recv: start
Send: N0 M110 N0*125
Changing monitoring state from "Detecting baudrate" to "Operational"
Send: N0 M110 N0*125
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N1 M115*39
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N2 M21*18
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor
```