OctoPrint/OctoPrint #2295: REST, files upload fails in cURL with a 411

mcuadros, 2017-12-17:
#### What were you doing?

I am creating a go API client, https://github.com/mcuadros/go-octoprint, and I am having some random EOF errors, so I decided to do it with cURL to fill a issue. 

I am using the docker octoprint images version 1.3.6 and the VIRTUAL printer.

My cURL command looks like:
`curl -F ‘file=@foo.gcode’ -vvvvvv -H "X-Api-Key: my-api-key"  http://localhost:5000/api/files/local`   

#### What did you expect to happen?

200 Ok

#### What happened instead?

```
* Connected to localhost (::1) port 5000 (#0)
> POST /api/files/local HTTP/1.1
> Host: localhost:5000
> User-Agent: curl/7.57.0
> Accept: */*
> Transfer-Encoding: chunked
> X-Api-Key: 48C135F39D9B4CF593C89F0B8424EA3F
> Content-Type: multipart/form-data; boundary=------------------------405e74f2e090a184
> Expect: 100-continue
>
< HTTP/1.1 411 Length Required
< Date: Sun, 17 Dec 2017 10:15:48 GMT
< Content-Length: 81
< Content-Type: text/html; charset=UTF-8
< X-Robots-Tag: noindex, nofollow, noimageindex
* HTTP error before end of send, stop sending
<
* Closing connection 0
<html><title>411: Length Required</title><body>411: Length Required</body></html>⏎ 
```

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint
```
Server:
 Version:      17.11.0-ce
 API version:  1.34 (minimum version 1.12)
 Go version:   go1.9.2
 Git commit:   1caf76ce6b
 Built:        Sun Dec 10 10:16:13 2017
 OS/Arch:      linux/amd64
 Experimental: false
```

#### Printer model & used firmware incl. version
VIRTUAL

#### Browser and Version of Browser, Operating System running Browser

cURL

#### Link to octoprint.log

https://gist.github.com/mcuadros/9e3e51287706112524b68cd0e1cd01a8

#### Link to contents of terminal tab or serial.log

None

#### Link to contents of Javascript console in the browser
None

#### Screenshot(s)/video(s) showing the problem:

None

ntoff, 2017-12-17:
`curl -k -H "X-Api-Key: redacted" -F "select=true" -F "print=false" -F "file=@0.3cube.gcode" "http://10.0.0.217:5000/api/files/local"`

works just fine here on a somewhat more regular install (not docker) using `curl 7.56.1 (x86_64-pc-win32)`

mcuadros, 2017-12-17:
Thanks!, I copy pasted the command from ... stackoverflow or something. Now works perfectly

closed 2017-12-17 by mcuadros