OctoPrint/OctoPrint #2296: X-Forwarded-For handling broken in flask-login < 0.3.0

bzed, 2017-12-17:
#### What were you doing?

Using basic auth using loads of x-forwarded-for headers.

#### What did you expect to happen?

Staying logged in at all times.

#### What happened instead?

Getting anonymous cookies + 401 errors at random times.

I run into the issue that is fixed by
https://github.com/maxcountryman/flask-login/pull/217
which is part of flask-login >= 0.3.0

#### Did the same happen when running OctoPrint in safe mode?

yes :)

#### Version of OctoPrint

OctoPrint 1.4.0.dev652+gbad3612e

#### Operating System running OctoPrint

debian stretch

#### Printer model & used firmware incl. version

doesn't matter

#### Browser and Version of Browser, Operating System running Browser

doesn't matter

#### Link to octoprint.log

doesn't matter

#### Link to contents of terminal tab or serial.log

doesn't matter

#### Link to contents of Javascript console in the browser

doesn't matter

#### Screenshot(s)/video(s) showing the problem:

doesn't matter

I have read the FAQ.

bzed, 2017-12-17:
btw, to reproduce the issue you actually have to use a list of changing X-Forwarded-For headers while keeping the First one the same - as it happens with load-balanced proxies or other funny networking stuff.

bzed, 2017-12-17:
Just realized that this is already part of the devel branch - thanks!;)

foosel, 2017-12-18:
As I'm wrapping things up for 2017, I'll take a look at this come January! 🎄

foosel, 2018-03-12:
Instead of updating this on the `maintenance` branch, I decided to apply this fix through a monkey patch for now.

The reason is that Flask Login changed its API in a backwards incompatible way by turning a bunch of methods into properties. On `devel` I've added a compat wrapper for that, but I don't feel comfortable right now adding that plus pushing a couple of dependencies up several versions (because Flask Login 0.3.0 also depends on newer Flask which brings its own rat's tail of changes) into the `maintenance` branch. The minimal monkey patch feels like the lesser evil for now.

I might revisit this in a future 1.3.x release.

In any case, the fix as committed above is now available on `maintenance` and will be part of 1.3.7. I'll of course have to roll it back on merge to `devel` ;)

commit a95407c8 (2018-03-12): Fix flask_login._get_remote_addr

We can't easily upgrade to 0.3.0 right now due to backwards incompatible
changes in that version (some methods became properties). So for now
we simply backport the fix in question through a monkey patch.

Fixes #2296
closed 2018-04-09 by foosel