OctoPrint/OctoPrint #2297: flask-login returns 401 by default, messing with basic authentication from proxies

bzed, 2017-12-17:
#### What were you doing?

I'm using a haproxy (as documented somewhere in the wiki ;)) with basic authentication.
Basically this works fine as long as I use autologin on octoprint side.
If not, I'm getting - mainly from plugins - some extra basic auth requests while not being logged in as the default behaviour of flask-login is to return 401 for requests (a plugin trying to talk to the api triggers that for example - I think....).

According to the flask-login documentation in my opinion the proper thing to do would be to add a login view that does nothing but returning a 403 (instead of the 401, which is the default).

```
unauthorized()
This is called when the user is required to log in. If you register a callback with LoginManager.unauthorized_handler(), then it will be called. Otherwise, it will take the following actions:

Flash LoginManager.login_message to the user.
If the app is using blueprints find the login view for the current blueprint using blueprint_login_views. If the app is not using blueprints or the login view for the current blueprint is not specified use the value of login_view.
Redirect the user to the login view. (The page they were attempting to access will be passed in the next query string variable, so you can redirect there if present instead of the homepage. Alternatively, it will be added to the session as next if USE_SESSION_FOR_NEXT is set.)
If LoginManager.login_view is not defined, then it will simply raise a HTTP 401 (Unauthorized) error instead.
```
#### What did you expect to happen?

Getting 403 errors instead of 401 from flask-login.

#### What happened instead?

401 errors triggered basic auth again as the browser can't make a difference between OctoPrint beeing funny and a broken BA.

#### Did the same happen when running OctoPrint in safe mode?

Actually I don't think so as nothing is shown which would trigger the issue. Didn't test it, though :P If plugins can trigger it, its buggy anyway ;)

#### Version of OctoPrint

OctoPrint 1.4.0.dev652+gbad3612e.dirty (yes, dirty, pull request incoming ;))

#### Operating System running OctoPrint

deb stretch

#### Printer model & used firmware incl. version

doesn't matter

#### Browser and Version of Browser, Operating System running Browser

doesn't matter (latest chrome)

#### Link to octoprint.log

doesn't matter

#### Link to contents of terminal tab or serial.log

doesn't matter

#### Link to contents of Javascript console in the browser

doesn't matter

#### Screenshot(s)/video(s) showing the problem:

doesn't matter

I have read the FAQ.

foosel, 2017-12-18:
As I'm wrapping things up for 2017, I'll take a look at this come January! 🎄

foosel, 2018-01-22:
Fixed on `maintenance`, soon `devel`, will be part of the 1.3.7 release.

commit b1f9e070 (2018-01-22): Return 403 instead of 401 on missing/insufficient credentials

Closes #2297
closed 2018-04-09 by foosel