OctoPrint/OctoPrint #2301: [Request] Exposing MQTT Publish Command function to Custom Control Editor (UI)

RJ-Make, 2017-12-18:
Hello Guys and Gales,
  I'm integrating OctoPrint's MQTT functionality into my home automation system and would love the ability to **easily** create a custom button inside the UI to send a MQTT command to my HA. 

I don't know enough to manually make this happen inside OctoPrint, so exposing the MQTT publish functionality for the say the Customer Control Editor plugin consumption would be fantastic.

jneilliii, 2017-12-18:
I'm working on a plugin to control tasmota devices using the [MQTT plugin](https://github.com/OctoPrint/OctoPrint-MQTT). You may want to look into that plugin as  well. You can find my early implementation over here...

https://github.com/jneilliii/OctoPrint-TasmotaMQTT

RJ-Make, 2017-12-19:
Looks interesting @jneilliii but after looking at the 'settings explained', I'm not sure how I can publish using MQTT?

jneilliii, 2017-12-19:
That readme is for my plugin.  I was just giving an example of how you could "roll your own" plugin using the MQTT helpers in @foosel's MQTT plugin.  Depending on what you want to do you might be able to use the existing publications from her plugin.  It publish's a lot of information from OctoPrint that you could "act on" in your home automation system.

jneilliii, 2017-12-19:
I just hacked together a quick example for you.  Install the [MQTT plugin](https://github.com/OctoPrint/OctoPrint-MQTT) first and then configure it to connect to your MQTT server.  Once that is configured then install this example plugin using the url below in plugin manager.  It will add a settings section MQTTPublish where you put in your topic and publishcommand (ie value) to that topic and save, and then press the play button in the navbar of OctoPrint and it should send that command to the topic configured.

https://github.com/jneilliii/OctoPrint-MQTTPublish/archive/master.zip

RJ-Make, 2018-01-02:
Sorry for the delay @jneilliii, busy with Christmas and all. That's pretty awesome what you've done here and it works as an example. I thank you very much. That said, I'm still not sure how to learn from it, not understanding it. 

Do you plan on developing a plugin that will expose the MQTT system for UI message sending?

jneilliii, 2018-01-03:
I'm not sure I understand your meaning. What are you trying to send? My example allows you to configure in settings what to send and sends it by pressing the button. Are you trying to have just an open text box to enter a topic in the main ui?

RJ-Make, 2018-01-03:
I'm trying to send messages for various HA commands. Turn On/Off Printer, Lighting, Filtration, etc.

jneilliii, 2018-01-05:
I've just released a new version to the plugin that allows for just that.  You can now add multiple topics/commands and configure the icons.

RJ-Make, 2018-01-06:
@jneilliii Now that is awesome! Every thing works except you can't see the icons. I've tried every UI combination all the way to a complete default setup. I've also tried leaving the icon field blank.

If you will indulge me, would it be possible to add:

1) A drop down for the commands?
2) Add a field for a description to be included with the icon?
 
See attached for Icon error. 

[image]

[image]

jneilliii, 2018-01-06:
That's odd, what about when the icon field is set to the default below? There are just way too many icons to populate into a select list. 

icon-play
or possibly
fa-play

RJ-Make, 2018-01-07:
Ok, so I'm embarrassed at the moment. I failed to read the header on the icon page and was using the unicode as the icon entry. Using the icon name works perfectly.  ;-{

... walking away feeling a bit dumb...

RJ-Make, 2018-01-07:
I'm really looking forward to seeing this plug-in evolve; I think it could be extremely useful to a lot of people. Do you plan on making MQTTPublish a full fledged plug-in, and if so is there a support page for your work?

anyway thanks again, I'm quite impressed with how fast you got this working, well done!.

jneilliii, 2018-01-07:
I could definitely release this to the plugin repository. You can submit to its github issues section for feature requests and support related needs.

RJ-Make, 2018-01-07:
Looking forward to it.

jneilliii, 2018-01-07:
I've submitted it to the plugin repository with a couple of other of my plugins.

RJ-Make, 2018-01-14:
Fantastic work on the latest version. Works perfectly!

Thank You!