OctoPrint/OctoPrint #2302: Nozzle Temperature Stops Updating at end of 1st print after updating printer FW (Repetier 0.92 -> Marlin 1.1.x)

akf-was-here, 2017-12-19:
Title:  Nozzle Temperature Stops Updating at end of 1st print after updating printer FW (Repetier 0.92 -> Marlin 1.1.x)

#### What were you doing?

I recently updated my printer FW from Repetier 0.92 (I think) to Marlin 1.1.x.  Prior to updating octoprint 1.3.6 had been operating normally.  After it appears to operate normally until the first print ends.  After the 1st print, the nozzel temp and nozzel target temp stop changing (i.e. they appear to flatline at the last temp recorded during active printing).  The bed temp and bed target temp continue to update acurately.

#### What did you expect to happen?

Nozzel temps in the graph continue to update.

#### What happened instead?

Nozzel temps in the graph Stop updatng.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

OctoPi 0.13.0

#### Printer model & used firmware incl. version

Tevo Tarantula
FW: Marlin 1.1.x

#### Browser and Version of Browser, Operating System running Browser

Win 10 running IE, Edge and Chrome 
FreeBSD running Mozilla Firefox

#### Link to octoprint.log

https://pastebin.com/xnArucYu

#### Link to contents of terminal tab or serial.log

https://pastebin.com/SerrZAUP

#### Link to contents of Javascript console in the browser

https://pastebin.com/ExfGMgkw

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[image]

akf-was-here, 2017-12-19:
between these lines in the terminal log:

Recv:  T:132.74 /0.00 B:53.88 /0.00 @:0 B@:0
Recv:  T:131.74 /0.00 B:53.85 /65.00 @:0 B@:127

I changed the bed temp to 65 on the printer and you can see that the temp graph (and terminal log) reflects this but the decreasing nozzle temp in the log is not tracked at all.

akf-was-here, 2017-12-20:
How do I know what information is missing so the gitissuebot doesn't delete this?

akf-was-here, 2017-12-22:
Disconnecting and reconnecting to the printer "resets" the temp graph functionality for the 1st print. 
 Manually stopping the 1st print before completing the print leaves graph functionality intact for the next print but then it's hung up after that print.

sbts, 2017-12-23:
@akf-was-here out of interest, rather than Disconnecting from the printer, what happens if you simply reload the Octoprint web page.
eg: by hitting F5 or ctrl-r

akf-was-here, 2017-12-23:
Refreshing the page doesn't change the temp data values as far as I can tell.  The Bed temp is still live and the Nozzle temp is still stuck.  I've accessed from several different devices (Computers, phones, tablets), operating systems (Win 10, Android, FreeBSD) and browsers (Edge, Chrome, Internet Explorer, Firefox) and the data all looks the same.

ntoff, 2017-12-27:
>How do I know what information is missing so the gitissuebot doesn't delete this?

I think it's the last bit, the "I have read the FAQ" line

akf-was-here, 2017-12-27:
Wow, that's kind of silly.  I updated it to include that line, maybe it'll get more traction.  Fingers crossed.

akf-was-here, 2017-12-28:
Recv:  T:220.94 /220.00 B:23.98 /0.00 @:84 B@:0
Send: M105
Recv: ok T:220.67 /220.00 B:23.98 /0.00 @:86 B@:0
Recv:  T:220.58 /220.00 B:24.02 /0.00 @:85 B@:0

akf-was-here, 2017-12-29:
Reconnecting my printer (so the temp graph is functional) and manually issuing T1 via the terminal also results in the graph locking up (as above).

Recv:  T:121.91 /0.00 B:22.58 /0.00 @:0 B@:0
Send: T1
Recv: echo:T1 Invalid extruder
Recv: ok
Recv:  T:120.80 /0.00 B:22.58 /0.00 @:0 B@:0

akf-was-here, 2017-12-29:
Tried to implement a fix in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py: 

                                if line.startswith("echo"):
                                        self._log("Handling echo response")
                                        continue
I deleted /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.pyc, restarted the server and verified the new pyc was built.  Issuing a T1 command results in:

Recv:  T:26.87 /0.00 B:20.78 /0.00 @:0 B@:0
Send: T1
Recv: echo:T1 Invalid extruder
Handling echo response
Recv: ok
Recv:  T:26.51 /0.00 B:21.29 /0.00 @:0 B@:0

but the graph's nozzle data is still flat lined at the Pre-T1 cmd value.

akf-was-here, 2017-12-29:
Issuing a T0 command seems to restore graph functionality.

akf-was-here, 2017-12-29:
Implemented another workaround (same spot as above):

                                if line.startswith("echo:T1 Invalid") or line.startswith("echo:T2 Invalid"):
                                        self._log("Handling echo response")
                                        self._sendCommand("T0")
                                        continue

Issuing a T1 command now results in:

Recv:  T:21.72 /0.00 B:20.74 /0.00 @:0 B@:0
Send: T1
Recv: echo:T1 Invalid extruder
Handling echo response
Recv: ok
Send: T0
Recv: echo:Active Extruder: 0
Recv: ok
Recv:  T:21.68 /0.00 B:20.43 /0.00 @:0 B@:0

Graph retails functionality after the command.

ntoff, 2017-12-29:
so what is issuing that T1 command? Do you have more than  one extruder configured in octoprint but not on your printer? Is the printer a shared nozzle (1 nozzle, 2 extruder motors?)

akf-was-here, 2017-12-29:
The T1/T2 is in the Gcode produced by Cura via user added gcode section to turn off both of my nozzles in repetier.  On switching to Marlin I've been unable so far to get my 2nd nozzle working in FW so I'm currently running the FW in single extruder mode until I can get around to messing with it again.

Shouldn't Octoprint just be updating it's graph based on the data being received from the serial print statements?  E.g.:

T:21.72 /0.00 B:20.74 /0.00 @:0 B@:0

means the temp of nozzel 0 is at 21.72 degrees and it's set to 0.00 degrees, while

T0:21.72 /0.00 T1:23.72 /0.00 B:20.74 /0.00 @:0 B@:0

means the temp of nozzel 0 is at 21.72 degrees and it's set to 0.00 degrees, means the temp of nozzel 1 is at 23.72 degrees and it's set to 0.00 degrees?  

My guess is that Octoprint is somewhere it's keeping track of the "Tx" commands being issued and assuming that once issued it's receiving data for a different nozzle for  "T:21.72 /0.00" serial print statements.  Maybe that's how some printers work?  Unsure, but if this is whats going on then maybe Octoprint needs to trap on an error response to the "Tx" cmd and not switch nozzle state in it's internal temp tracker.

I could try to hunt down this nozzle tracker code myself, but I am not a python guy and I'm not really familiar with this code base.

foosel, 2018-01-11:
I'm confused... Your printer has multiple extruders, correct? Why does your firmware claim that T1 is an invalid tool then?

Temperature parsing is a bit tricky since the report contents vary greatly across firmwares and even firmware variants. See [the comments in this code section](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/comm.py#L3274-L3360) for a small idea.

Before I can dive deeper into this I need to know why your firmware is only reporting on extruder temperature here and refusing to switch tools. From your logs it looks like your firmware doesn't know it has more than one extruder, and then you shouldn't configure those in OctoPrint either. `T1` definitely shouldn't throw an error and - unless you are running a shared nozzle setup here - your temperature reports should contain temperatures for all of your extruders, not just one. So we first need to figure out what's up there on the firmware side before it can even be determined what's going wrong here with OctoPrint's graph and how best to solve it.

akf-was-here, 2018-01-11:
My printer has 2 physical nozzles, both of which were functional in Repetier but were not working correctly after my initial Marlin build. As a result, I rebuilt Marlin with only 1 nozzle enabled until I can get some time to go back and figure out how I had the Marlin FW mis-configured for 2 nozzels.  

As a part of the manually written Ending  G-Code in my slicer it's explicitly switching to each nozzle (T1, T2) and setting the temp to 0 (at one point it had been leaving my nozzles running after print completion which seemed dangerous). In Repetier this has been working just fine for a year or so. In Marlin, with only one nozzle enabled, it's not.  Probably for good reason since only once nozzle is enabled in the FW build.  This G-code does execute fine and leave the printer in a good state though, it's only Octoprint that gets confused.

I can fix this on my side by editing my end code, or by implementing the fix I outlined above (which is what I currently have in place) if you don't want to fix it for everyone. This is kind of a corner-case but it seems like you could gracefully handle failed attempts to switch to a different nozzle. It may be, though,  that I'm just oversimplifying this issue when extrapolated to all FWs as you indicate is the case for temp parsing.

foosel, 2018-01-12:
Ok, I now see why things are as they are. 

Your modification in principle makes sense, needs a bit more work around the etches though. And I'll also need to figure out why you don't see *any* temperature logging on the graph anymore - from your setup I'd have suspected OctoPrint to simply wrongly log the temps of T0 for T1 after the switch. So I'll have to look at what's up there (and I sincerely hope that isn't a side effect of the bending back and forth to be compatible to the zoo of temp reports out there).

Until I've fixed that though, you might want to take a look at [this minimal plugin](https://gist.github.com/foosel/5ede72de0185fad7a6f092d878ed8430) I just quickly whipped up. It prevents any T commands above T0 to be sent to your printer in the first place. It can be installed on OctoPi by SSHing into it, then:

    cd ~/.octoprint/plugins
    wget https://gist.githubusercontent.com/foosel/5ede72de0185fad7a6f092d878ed8430/raw/c47d8b86517b6bfada59bc6b77923bcf00703562/dont_send_t1_and_up.py
    sudo service octoprint restart

commit 551d6aab (2018-02-26): Detect invalid tools according to firmware

Also don't send invalid tool commands (based on printer profile and
former responses from the firmware on the ongoing connection).

Closes #2302
closed 2018-04-09 by foosel