OctoPrint/OctoPrint #2303: Gcode Visualizer - Wrong aspect ratio

Thisismydigitalself, 2017-12-19:
OctoPrint 1.3.6 running on octppi 0.13.
My Printer's physical bed size X400 Y250. My Octoprint Printer profile reflects that.
Still, Visualizer showing square grid which does not represents the actual aspect ratio.

ntoff, 2017-12-19:
zoom out, the bed should be a rectangle shape. The squares stay square because they're measurements (10x10mm I think), but the overall shape should be a rectangle.

Thisismydigitalself, 2017-12-19:
Zooming out reveals another issue: bed cut in half. missing the right side of the bed. 
[image]

ntoff, 2017-12-19:
that's weird, it doesn't do that for me. 
If you could please:

* Try it in safemode (system menu -> restart in safemode)
* Post logs (found in octoprint's settings menu -> logs
* take a screenshot of your developer console (F12) to see if there are any javascript errors

Just be aware it might be after christmas / the new year until someone is able to take a look into it.

Thisismydigitalself, 2017-12-19:
Thank you. no pressure. enjoying OctoPrint regardless :) but it will be nice to solve this one out.
Running in safe mode did not change the outcome.  i will see if i can upload some files later on.

[image]

Thisismydigitalself, 2017-12-19:
Counting the number of tiles it is clear that OctoPrint for some reason loads a 200 x 200mm printbed grid- just like the default printer defaults. this means my Octoprint will ignore my custom printer profile's settings. This might give a clue what goes on under the hood.

Thisismydigitalself, 2017-12-19:
OK!
My previous post made me think my issue is somehow related to OctoPrint printer's Default profile so what i did was to edit the default printer profile and this fixed my problem. can this be it?
i was under the impression that creating new printer and define it will create a new profile for octoprint use but apparently this is not the case. one needs to edit the default printer profile to get it to work.
Not intuitive.  What am i missing?

ntoff, 2017-12-20:
>i was under the impression that creating new printer and defining it will create a new profile for octoprint use but apparently this is not the case. one needs to edit the default printer profile to get it to work.
Not intuitive. What am i missing?

Yep, looks like a bug, rather than pulling the details from the profile that's currently set as default, it jut pulls it from original "Default" profile.

Thisismydigitalself, 2017-12-20:
Wow, - Could I have missed that?
Then again, why one needs different profiles to begin with if these are not separated machines? kind of confusing. I do remembered to check the yellow asterisk symbol next to my printer's profile so i was hoping it will load it as default. second thing i jut checked - when you change printer profile under connection tab - you must refresh your browser to see the change. changing the profile and reconnecting must also refresh one's browser. this is not the case. Gcode Vizualizer will NOT change it's grid if you only change profile and reconnect - you must also manually refresh / reload the page.

Thanks for your help ntoff!

ntoff, 2017-12-20:
The profiles are there if you want to connect different printers I suppose. OctoPrint can be run on just about anything, including the latest threadripper based PC's, to which you could probably connect 30 printers at once or something ridiculous. It's not limited to running on low end hardware that only supports a single printer at a time.

You're not alone in missing it, I totally forgot that option existed as well since I only ever use the default profile that's bundled (hence the edited comment). Perhaps a usability improvement could be up in the state panel, it could list the current profile in use. Something like **Connected (profilename)** might actually make sense, especially if someone does happen to have multiple printers connected to one machine running octoprint. 

Maybe an extra line for the profile name that's in use 
[image]

Thisismydigitalself, 2017-12-20:
Different printers can have different connection settings and/or protocols. OctoPrint Printer's profiles are missing these. I can understand running multiple instances of OctoPrints on the same Pi (i chose to have pi per printer) but if i don't really understand the use of printer's profiles unless these are shared between instances.

closed 2018-01-03 by GitIssueBot